BrobstonCreations/yet-another-smart-vent #1: The vent shell is too big

jorgecis, 2022-08-18:
I notice that for the vent 6x12 you split the bigger parts but not the shell, and is not possible to print it using the Prusa (Not the XL that is not available yet)

Can you split the shell?

TonyBrobston, 2022-08-18:
Yep, that's the plan. 

I've got a note in the readme and a game plan in my head of how to make it happen, should be available in the next few days as long as testing goes well.

Thanks for your interest, let me know if you have any other questions.

TonyBrobston, 2022-08-19:
First iteration didn't go so well. Working on a second iteration today.

TonyBrobston, 2022-08-20:
[image]
Couple more iterations done. This is pretty solid; I'm surprised this even works. I'll test print a 2x12 in the next day or two. Assuming that works I'll update the 4x12 and 6x12 and test those.

Vader699, 2022-08-20:
Awesome.  I was looking into the 4x12 and wondering what you thought would be the best way to split the shell.  

And it's kind of too bad having to print the louvers diagonally so they fit on my bed.  going to have to do them one at a time it seems.

TonyBrobston, 2022-08-20:
All my vents are 4x10 and 6x10, so until less than a week ago, that's all that existed.

This snap together concept is still sort of blowing my mind. It audibly snaps when going together.

Yeah, there's probably a way to make the louver a two-piece design but I think the cons heavily outweigh the pros. I personally have two magnetic print beds, so I glue stick my spare while my primary is running and do a quick swap and keep the printer going. But, obviously I have to be present to do it 🤷‍♂️.

TonyBrobston, 2022-08-21:
I'm test printing a 2x12 now.

TonyBrobston, 2022-08-22:
[image]

The shell is printed. It snaps together. It's a bit flimsy until you put the two screws in. I think once the top is printed and installed it will gain enough rigidity to be viable, but I can only speculate until then. I have two longer louvers and end-caps printing now; then two vent top halves after that.

TonyBrobston, 2022-08-22:
I'm not loving this snap together design.

I'm probably going to iterate on having the vent shell side overlap; the width is 2mm, so I'll print two 1mm walls that overlap. I'll design for four M3 screws to clamp them together. I think this will give more rigidity.

My fear is putting out something that may have issues down the road; it seems like less work to get it right from the start then to apologize for a failing design and try to remedy it later.

TonyBrobston, 2022-08-23:
I tested this out, works great. Next I'll switch the 4x12 over to this two-piece design.
[image]

Vader699, 2022-08-23:
Nice looking pretty good!

TonyBrobston, 2022-08-23:
Thanks!

Slowly but surely and a bit of luck 😄 .

TonyBrobston, 2022-08-24:
I've got the 4x12 shell printed. Working on the top now, then 2 more louvers. After that I'll know for sure if this works. The split shell is up for 2x12, 4x12, and 6x12; but I've only tested the first.
[image]

TonyBrobston, 2022-08-24:
I tested the 4x12 and it's fantastic. The vent top and specifically the M5 screw give the shell and snap joint way more rigidity than I expected. Essentially the side-to-side flex was bad, but when attaching the top 99% of that goes away. I'll chalk that up to dumb luck.

My printer needs some maintenance, so I won't be testing the 6x12 for awhile.

[image]
[image]

TonyBrobston, 2022-08-27:
I'm going to go ahead and close this issue. I'll probably post back once I test the 6x12, but it is going to be awhile. I'm 99% confident it's good to go.

If anyone has any other questions feel free to reopen this issue or open a new issue.

closed 2022-08-27 by TonyBrobston