BrobstonCreations/yet-another-smart-vent #2: Daughter board on the D1 mini?

Therealjosephchrzempiec, 2022-08-19:
Hello, What is that daughter board on the D1 mini? There is nothing in your information to show or tell what is is.

Joseph

Therealjosephchrzempiec, 2022-08-19:
But I do have a question on the power shield board. How come most of the sites and videos I saw no one is using a Power shirld and you are?

TonyBrobston, 2022-08-19:
The short answer is, voltage drop.

Initially I I've designed all of this stuff to fit my needs. As others describe their needs things will likely change. 

I knew from the beginning that I would run FPLP (Fire Power Limited Plenum) wire for power/ground from my utility closet where my furnace is, through the ducting, to the end of each ducting run. Next to the furnace I have a [power supply](https://www.amazon.com/gp/product/B0146IAXYO) powering each run. As a side note, at some point I'll release a power supply enclosure cover to work with something like [this](https://www.holidaycoro.com/product-p/1308.htm) and a 24v power supply. Voltage drops with greater resistance and the longer the wire the higher the resistance, a 7-24v power shield that provides 5v to the ESP8266 D1 Mini seemed like a great option. The power supply sends 24v, but at the end of the ducting run I'm sure there's less than 24v, but greater than 7v.

Yes the ESP8266 D1 Mini only needs 5v. So you could forgo the power shield if you want to save a few dollars per vent, if you were to use the micro-usb port to power things. Though I'll need to modify the lower enclosure to hold only the D1 Mini. Currently there is a piece on the enclosure lid that holds the D1 Mini from the top, but the bottom is held by the power shield's female DC connector.

Let me know if you have any other question, I'm happy to answer them.

Therealjosephchrzempiec, 2022-08-19:
Hello, Thank you very much for the information. That explains a lot.

TonyBrobston, 2022-08-19:
No problem, happy to help.

Feel free to open another issue if you think of more questions.

closed 2022-08-19 by TonyBrobston