BrobstonCreations/yet-another-smart-vent #3: Any chance of a 3 x 10 model?

reuter, 2022-08-19:
Canada seems to have a not as common register size. All the smart cents on the market are 4 x 10, but there's no 3 x 10 sizes. 
I'm happy to print and test, but my CAD skills leave much to be desired.

TonyBrobston, 2022-08-19:
Yeah, that's totally do-able. I added it to the list in the readme, it should be pretty straight forward; I'll get to it when I have some time; probably in the next week.

reuter, 2022-08-19:
Wow! Awesome!

TonyBrobston, 2022-08-20:
https://www.printables.com/model/262311-yet-another-smart-vent-3x10

3x10 is now uploaded. Everything seems right to me, but it all needs printed and tested. If you happen to get to it before me, please report back with your findings; I would be surprised if I got anything wrong, just because I made and tested a few variations and have had a 100% success rate.

Let me know if there's anything else I can do for you.

closed 2022-08-20 by TonyBrobston

reuter, 2022-08-22:
Printed, and everything looks awesome. Will try out the electronics when I get back from this trip. 

[image]
[image]

TonyBrobston, 2022-08-22:
Looks good!

TonyBrobston, 2022-08-22:
When you have this fully assembled would you mind adding some more pictures here and would you mind if I put them as the primary pictures on printables? I may eventually print a 3x10, but I've got a lot of other things that need testing, so it's either going to be awhile or not at all.

reuter, 2022-08-22:
Of course!
Once it's up and running, I'll do videos and all 😀
Will be a couple weeks. I need to spruce the components first. Having trouble up here in Canada.

TonyBrobston, 2022-08-22:
Awesome!

I imagine there are a lot of options for everything except the Batan B2122, though I believe that can be ordered from a few places, but ultimately Adafruit seems to be the one that is having them made.

Nuetron06, 2022-08-22:
Maybe a possibility of using another servo?

reuter, 2022-08-22:
Found these

Miuzei 5 pcs SG90 Servo Motors Kit Mini SG90 9G Servo Motor for RC Robot Helicopter Airplane Car Boat Remote Control Blue https://a.co/d/hhqTej5

DWEII 10pcs DC Power Shield V1.1.0 for wemos D1 Mini Development Board DC Power Supply Screen Module https://a.co/d/0GdAt8L

CANADUINO 3 x WEMOS D1 Mini ESP8266 4MB Wi-Fi Module - Compatible with Arduino and Lua https://a.co/d/0jfOPrU

Hoping those will work.

TonyBrobston, 2022-08-22:
Just a heads up on the SG90's and MG90's, I've experimented with both of these a ton and from my experience they are more effort than they are worth. At a minimum, be sure you get TowerPro and that they're authenticate and not fakes.
https://vorpalrobotics.com/wiki/index.php/Tower_Pro_MG90S_Vs._Clones

Also, these different servos come with different sized wiper arms and slightly different mounting points (a difference of less than 1 mm, but enough to mess things up).

What are you seeing when you try to order the Batan B2122?

TonyBrobston, 2022-08-22:
To elaborate a bit. 

The counterfeit MG90's claim to have metal gears, but only the front gear is metal, the rest are plastic.

The counterfeit SG90's are plastic gears only.

It might have been from stressing the servos too much during development, but I burnt up and or broke teeth off on multiples of both SG90's and MG90's.

Also these are 3 wire servos, so if you did manage to go that route,  you have to open to the servo to add a wire to get signal from the potentiometer. This also means I would have to publish multiple servo levers to account for different servo wiper arms and multiple enclosure tops/bottoms to account for the mounting points being off by 1mm or so.

I do have one authentic TowerPro SG90, the quality seems to be pretty good; it's likely that this would work under normal operation.

In the end I was more afraid of producing/recommending low quality products and ending up regretting it when people have all kinds of issues. At a minimum I would recommend using the Batan B2122, you can definitely go with counterfeit, cheap, knock-offs for everything else, as I did with success.

reuter, 2022-08-23:
Found these. Any of these OK in your opinion?
https://www.pishop.ca/product/micro-servo-sg92r/
https://www.pishop.ca/product/sg90-180-degrees-9g-micro-servo-motor-tower-pro/
https://www.pishop.ca/product/sg5010-towerpro-torque-coreless-servo-180-degrees/

TonyBrobston, 2022-08-23:
Unfortunately from what I've seen with the SG90's, I would have to make changes to the design files to put the mounting points for the servo at the right distance from the lever that it turns, as well as modify the lever to allow a different sized wiper arm (which varied between SG90's), then release multiple versions so that things are compatible with the recommend servo and the SG90. Then you would need to open the SG90 up and add a wire to give the D1 Mini signal from the servo's potentiometer. Then on top of that, you may have issues down the road with the SG90; but maybe not if it's an authentic TowerPro.

IMO, this all boils down to being worth figuring out how to get a Batan B2122 and save us time and potential frustration.

The main place I'm seeing online to buy the Batan B2122 are:
https://www.adafruit.com/product/1450
https://www.digikey.com/en/products/detail/adafruit-industries-llc/1450/5154665

Are they not offering to ship to Canada?

I have 15 of these motors on order from Adafruit. I'm working on getting some stuff together to offer electronics kits in the somewhat near future.

TonyBrobston, 2022-08-23:
As for the SG5010, I think there are similar issue; other than the quality is probably good. I've also coded specifically for the Batan B2122 with is a 120 degree servo, whereas some of the other options are 180 degree or more.

reuter, 2022-08-23:
I'll keep digging to find a place to get these exact ones.
I just discovered digikey.ca which has the servos. Just need to track down the D1 boards now.

Thanks for the help and sorry to be a pain (this is my first foray into this area of the hobby)

TonyBrobston, 2022-08-23:
Don't apologize, you're not being a pain.

I appreciate you diving in; my goal is for people to use this stuff, so in a way you're helping me reach that goal.

Sounds good on the digikey.ca. I'd say probably any ESP8266 D1 Mini on Amazon that has decent reviews and is like $2-5 a piece and looks roughly the same as what I'm using in the [Electronics Assembly](https://github.com/TonyBrobston/yet-another-smart-vent/blob/master/docs/ELECTRONICS_ASSEMBLY.md).

TonyBrobston, 2022-08-23:
I just realized it's hard to see the D1 Mini under the Power Shield.
[image]

TonyBrobston, 2022-08-23:
Sounds like a plan.

Also, just a heads up that sometimes when the calibration cycle runs on startup it prematurely thinks it finds the endstop. When I set the vent down on it's top it seems to find it correct nearly every time, but if I'm holding it, it sometimes is premature. I have a game plan to fix this, I just feel that getting the 12 inch vents split so people can print them is more pressing. There are a couple ways I could fix this:
1. I'm currently checking how far the servo has turned after each 1 degree move, when it basically stops moving, I know I'm at the end stop. Instead of comparing the last position against the current position, I could compare the last two positions, or last three positions (you get the idea) and this would make it less sensitive.
2. I know how many degrees we have turned at any given point during calibration. I know that looking in Fusion 360 the vent can turn 87 degrees, so if the endstops are less than 87 degrees apart, the calibration stopped prematurely. At which point I could tell it to recalibrate, but this could result in an infinite loop if the second or third or fourth attempt also fails.
3. Another option is a variation of the last one. Let's say there is 50 degrees between endstops after calibration, we know 87-50=37; so the max possible travel on either side of the premature endstops is 37, but is likely less. So, I could repeat open, then close calibration with a sensitive strategy for finding the end stop and then once we've hit 87 degrees in total, we know we've succeeded.

I have this noted in the readme as: "Make open/close calibration slightly less sensitive." I'm hoping to get to this after finishing the 4x12 and 6x12 vent splits and creating a 6x8. At some point I hope to have an way to update the firmware on the vents without needing to connect a cable. The only two options I'm aware of are: 1. ArduinoOTA (if we stick with coding the whole solution) or 2. ESPHome (which I'd like to do a POC on to determine viability, this would significantly decrease the complexity of this project because I can essentially delete wifi/mqtt setup and get that for free from ESPHome's setup).

Nuetron06, 2022-08-23:
How about this for servo?
https://www.mouser.com/ProductDetail/DFRobot/SER0046?qs=vHuUswq2%252BsyGXvdy%2FXoNTA%3D%3D

TonyBrobston, 2022-08-24:
Those servos look like they are worth experimenting with. I like that they're like half as expensive as the others.

I'll see about ordering one and see what the differences are. 

Whatever we do, we probably need to find a servo and stick to it, because supporting many is likely going to cause problems and confusion. I can't say for sure though.

TonyBrobston, 2022-08-24:
I ordered one servo. $15 after shipping, ouch. But... $6.90 is a lot less than $14.95 or $13.46 for the [Batan B2122](https://www.adafruit.com/product/1450). Obviously there's shipping costs to consider too, but if buying in bulk, it's probably free shipping.

I'll get this servo tested when it shows up and report back; shipping was up to a week and then it'll take me at least a few days to get to experimenting.

Nuetron06, 2022-08-24:
One of the things I did notice after researching a bit on this model, it might be 3mm taller.

TonyBrobston, 2022-08-24:
I should probably measure the Batan B2122, but I'll just compare it to the SG90, the SG90 is slightly larger in a few of it's dimensions.

- 12.2mm should be fine for the width, I just need to make the mount point wider; there is plenty of space.
- The height is 33.4mm, I have 34.5mm of space available.
- The depth from output shaft to the back of the body is 32.5mm (28.5mm+4mm); this is probably the most concerning, because the SG90 is 30mm. 2.5mm is a decent amount of space. It's also hard to say how far the wiper arm will stick out from the servo output shaft. One way or another I'll make it work. On my second iteration, months ago, I made a relief in the electronics enclosure wall to fit my authentic TowerPro SG90; I can do that again or actually leave an opening for the motor to consume all that space.

[image]
[image]

TonyBrobston, 2022-08-27:
My DMS-MG90-A Servo delivered yesterday. It will be a few days before I get to testing it, but it looks super promising. Definitely has metal gears and four wires, so I can't see why it wouldn't work. I'm also thrilled that it's less expensive as I would like the cost per vent to be $10-15 lower and this accounts for $6.56, so that's a good start.

I'm currently learning ESPHome and converting this project to that. Hopefully I'll have that worked out in the next few days and can then test this.

TonyBrobston, 2022-09-03:
Testing the DMS-MG90-A currently. Finishing printing the last part to put a full vent together to see where I'm at.

Couple things worth noting. The DMS-MG90-A turns the opposite of the Batan B2122; not a big deal, just a small code change.  The DMS-MG90-A can turn 270 degrees and the Batan B2122 can turn 120; again, not a big deal this just effects how fast the servo turns in ESPHome based off the `transition_length`. Currently I have it set to 3 seconds; so I'll probably scale it to 6.75 seconds. The DMS-MG90-A does not seem to have ball bearings (or maybe any bearings?), so operation might be a little less smooth (or something?) than the Batan B2122.

But... $14.95 or $13.46 vs. $6.90 is hard to argue with 🤷‍♂️ .

TonyBrobston, 2022-10-16:
I released support for the DFRobot DMS-MG90-A earlier today. This seems to be a less expensive, but comparable servo to the Batan B2122.