BrobstonCreations/yet-another-smart-vent #4: how well is this working for you?

wjcarpenter, 2022-08-20:
Not really an issue. Just a question. This is a well-organized project with a lot of thought behind it. It looks like you are reasonably far along. I am curious if you are actually using this to balance your HVAC airflow yet.

I'm not "an HVAC guy", but I've been trying to figure out an economical (or completely DIY) way to balance my single-zone system so that my upstairs rooms are similar to my downstairs rooms. My impression (from Internet searching and reading) is that the "close off some vents" schemes are trickier and less desirable than the "booster fans in ducts or registers" schemes, mainly because the increased static pressure is harder to manage. But I'm no expert, and I wonder how your system works in practice.

Are you far enough along to be able to draw any conclusions?

TonyBrobston, 2022-08-20:
It's working pretty well. To try and draw a better picture I'll explain some of what I've seen. 

I live in a three-level-split, so I have two zones (which does not have any traditional zone stuff going on). One zone is the main floor, which has four 4x10 vents; two rooms with two vents each. The other zone is the basement and upstairs. The basement has one room with three 6x10 vents and the upstairs has four rooms with one 4x10 vent each. Also, my basement is only half below ground.

It is currently 68-95 F here (Des Moines, IA). From what I can tell, the upstairs is the hardest to keep cool when it's 90+ F (mid-day on a hot day); especially my office (smallest bedroom) and our master bedroom (which has a sunroom off of it, hot air leaks from the sunroom to our master bedroom). The short answer is, I can now get both of these rooms down to at least 70 F; but tend to prefer 72/73 F.

I wondered for the longest time why my office would get so hot. Part of that is two PC's, two large monitors, lighting, my 3d printer, body heat, southern exposure, etc. But also, the ducting for this room is second in line; counting ducting runs for this zone, my office has another five ducting runs after it (two upstairs and three in the basement). When all 6 vents in this zone are open, it seems like my office is getting less than 1/6 of the airflow. When all or a few of the five ducts past my office are shut, my office gets substantially more airflow, and as a result, cools down. I can now get it to be colder than I like, even with my computers and printer running.

On the other hand, I would prefer my basement at 70-72, but it tends to sit at 68-71, even when the basement vents have been closed for days. I attribute this to being half below ground, having rooms that are cooled above it, and possibly some cool air leaking from the HVAC (I've noticed if I feel around the main plenum above the furnace, there are small air leaks). In theory I could make [mqtt-hvac-vent-control](https://github.com/TonyBrobston/mqtt-hvac-vent-control) smart enough to compensate by pushing some heat to the basement, but I'm not sure that's something I want to do. But, this sort of concept could make sense if the situation was inverted and you had a room creating a lot of heat (like a server room/closet); it might be winter, but maybe you want to push cold air to one room to keep server temps down, though I have no idea the implications of running A/C in the winter 🤔 .

The vents do open/close often. They aren't silent; I'm not sure it's possible to have a motor that is silent? As I explore battery options, I may experiment with some [sleep mode options](https://docs.arduino.cc/learn/electronics/low-power), which will inherently cause the vents to open/close less often.

As for the static pressure, from what I've gathered there's really only two options:
1. A variable speed blower, which would allow you to match CFM to the system based on how many vents are opened or closed. However, I'm not sure how this accounts for blowing enough air across the coil to keep it from freezing; I also don't know if that's an actual thing, but have heard others mention it.
2. Installing a [Static Pressure Regulating Damper](https://www.zonefirst.com/product/sprddd/) between the main return and the main plenum. This would still move the required amount of air across the coil, but I'm not sure if recirculating cold air would cause the coil to get too cold or not. From what I saw in this video, it does not seem like a concern. https://www.youtube.com/watch?v=WSJUQ2qEb3w
3. Installing some sort of diverter. Similar to #2, but not recirculating. This seems like the safest, but not exactly the most efficient.

*If anyone else has suggestions on how to deal with the static pressure issue, please comment*

wjcarpenter, 2022-08-20:
Thanks for the detailed response, Tony. (BTW, you are probably already familiar with this commercial product. It's too expensive and proprietary for my tastes. The idea is similar, though. https://flair.co/)

A few weeks ago, I had a heat pump installed, and a new furnace with a 10-speed fan to go with it. That upped my interest in finally solving the "upstairs too hot/cold" problem. Previously it was only "too cold in winter". Now it's that plus "too warm in summer". The ideal system would deal with that while also taking into account the fickle nature of family member preferences. 

I've been looking at all the commercial products I can find (which for some reason seem to be mostly unavailable at this time) and also considering DIY options (I don't have access to ducts except in the basement, so I'm mostly thinking about things in the affected rooms). My major conclusion so far is that HVAC is more complicated than I thought, and I also wonder if the average "HVAC guy" actually understands all that stuff or just wings it with rules of thumb. :-)

I had initially more or less rejected the idea of closing off registers (as in the Flair system and in yours) because of cautions about it from strangers on the internet (who mostly say it just doesn't work). OTOH, it's hard to argue with success. There are also concerns about it's effects on the health of your HVAC system, which you seem to be pretty aware of. This project and your experience with it are leading me to consider it again. The reason I concentrated instead on register booster fans is because I figure they will always reduce static pressure unless they are grossly mis-designed. But what the heck do I know?

Thanks again for your response and for your work on this project.

wjcarpenter, 2022-08-20:
Here are a couple of articles about controlling static pressure that I found interesting. They are mostly about HVAC in commercial buildings, but I think most of the principles apply.

- Building Re-Tuning Training Guide: AHU Static Pressure Control https://buildingretuning.pnnl.gov/documents/pnnl_sa_84187.pdf
- Technical FAQ: HVAC: Excess Static Pressure Relief https://www.homeclimatecontrol.com/faq/technical-hvac-excess-static-pressure-relief

TonyBrobston, 2022-08-20:
I would also mention I'm a software engineer, not an HVAC professional. So, proceed with caution.

However I would like to emphasize even more heavily that there are a lot of dismissive people in the world. So when "that will burn your house down" comes up, I "trust but verify".  I'll challenge the person to explain why. Then seek a solution that remedies the concern, like an engineer does.

There's no way these systems are so complicated that we can't comprehend them.

I do agree, I think a [booster fan concept](https://mysmartcocoon.com) could side-step the static pressure concern. But, I have a hard time believing this is the only way to do it.

kamermans, 2022-08-20:
This seems like a cool (excuse the pun 😂) project!   How about making the louvers adjustable.   If you define a desired total airflow/volume (static pressure is related to this if I understand correctly) of like 80% (20% closed), then you can compute offsets for all the vents in order to redistribute the air while maintaining that percent.   For example, if you have 10 vents in one zone that are open to 80% and you need more cool air in one room, change that vent to 100% and the other 9 to 77.8% to compensate.

Of course there is some sort of mathematical curve to describe the angle of the louvers vs the airflow, but I'm sure that idyllic to hard to determine.

TonyBrobston, 2022-08-22:
One of the next code changes I plan to make to the vent is: `Add the ability to move the vent to a specific position between 0% and 100%.`, which is documented in the readme.

This will then enable more intelligent control in [mqtt-hvac-vent-control](https://github.com/TonyBrobston/mqtt-hvac-vent-control). Right now I envision changing closed from 100% shut to something between 70-90% when the number of vents shut is nearing 100%; not sure of an exact number yet. I think this in combination with a static air handler lowers my concerns about static air pressure and not getting enough air to pass across the coil to keep it from freezing.

TonyBrobston, 2022-08-22:
I'm going to close this for now, if you have any more questions feel free to reopen or open a new issue.

closed 2022-08-22 by TonyBrobston