BrobstonCreations/yet-another-smart-vent #5: Request: Split-up version of vent for Prusa Mini

KTibow, 2022-08-21:
Has anyone tried to split up the vent into multiple parts? I want to print the 4x12 vent, but just rotating the parts won't get them to fit inside the print bed.
[image]

TonyBrobston, 2022-08-22:
I'll put some thought to this. It's certainly possible one way or another.

I'm currently wrestling with how to maintain all these variations long-term and avoid creating confusion about which parts are needed depending on your printbed size.

TonyBrobston, 2022-08-24:
As of right now, I'm leaning towards designing for printing on a regular Prusa.

I've spent the last few days focused only on different vent variations and in some cases splitting vent tops and shells to be printable on the i3 MK3S+. I'm currently working on this by myself; it's likely that others will help test and maybe at some point someone will help with software development. With this in mind I need to be smart with my time, so for now I'm going to move on to other things. Some day I may support the mini, but not for the foreseeable future.

It pains me to say that, because I don't want to leave anyone out of the fun; but I think the reality is that I don't have the time, at least yet.

Let me know if you have any other questions.

KTibow, 2022-08-24:
Not a big problem, it's not like I really need this. I could probably modify it if I wanted to.

TonyBrobston, 2022-08-24:
Gotcha, that seems reasonable.

Another thing to consider, if you're looking at another Smart Vent is the cost difference. You could almost pay for a Original Prusa i3 MK3S+ 3D printer with the cost difference. This probably makes more sense for people that don't yet have a printer. Also the features between those other Smart Vents and this one is not a 1:1, at least yet.

I'm going to close this issue for now, feel free to reopen or open a new issue if you have any more questions.