BrobstonCreations/yet-another-smart-vent #6: Release original design files?

tommck, 2022-08-21:
This is a great project, but for those of us who have unusual vent sizes, would you consider releasing the original design files so we can customize things ourselves?

TonyBrobston, 2022-08-22:
As of right now I'm not planning to release the design files. It's possible that this could change. Part of the reasoning is around licensing and the potential to offer kits or assembled vents at some point down the road. I've spent a lot of type on this system and feel that it's pretty generous that I'm offering what I am, at no cost to the users. I feel that opening these files up will likely be like letting the genie out of the bottle and not being able to get it back in, even if a license is there to protect from that. I hope that being transparent is helpful, while I realize not releasing the files is not 🤷‍♂️ . 

What I would prefer to do for now, is understand the needs of those that want to use this system and do my best to address those needs. Would you mind sharing the vents that you have and we can go from there?

Another option is to design a vent. [This repo is licensed](https://github.com/TonyBrobston/yet-another-smart-vent/blob/master/LICENSE) in such a way that you can use it  in a different vent.

closed 2022-08-23 by TonyBrobston

hllhll, 2024-05-19:
Hi @TonyBrobston  Can you please provide a purchasable option for a 20x40cm (~7.8x15.7") or maybe allow buying the sources in "coffee cup" price ? :);  
I Actually have NO WAY of getting a smart vent for the local vent sizes we use here (above)

Just for an idea, My friend is building a DIY  game arcade gun, that being said, the maker does make some change on it, he is releasing the models and documentation only with a small amount of $