BrobstonCreations/yet-another-smart-vent #7: what type of power wiring?

wjcarpenter, 2022-08-21:
I'm curious what type of wire you used for the power runs. You described running 24v through the air ductwork. That may or may not meet electrical codes (zillions of factors there, and I'm not an electrician). Assuming you're happy with the code situation, what's the best wire to use?

- You probably want to use something plenum-rated so that it doesn't make things worse in the event of a fire (spreading the fire or emitting toxic fumes).
- It's probably easy to find some wire that won't be directly melted by the temperatures when hot air is flowing (except for extreme heat right near the heat exchanger). I'd be concerned about it degrading or becoming brittle over the course of a few years.

Did you do any research about this before running your wires?

dvsxutact, 2022-08-21:
I was curious about this as well! I haven't finished reading through everything yet, but i was thinking about either a small battery pack, or just a DC power jack on the vent itself.

TonyBrobston, 2022-08-22:
I ran 18 AWG FPLP (Fire Power Limited Plenum) wire.

[According to this link](https://www.foxvalleyfire.com/2020/11/fire-alarm-cable-matters): "FPLP (Fire Power Limited Plenum) Plenum Fire Alarm Cable is used in horizontal overhead installations such as, duct work or other space that is used for the flow of environmental air."

The wire I rand is the "Fire Alarm and Security Control: Shielded and Non-shielded" Cable Type in this picture:
[image]

So, rated up to -4 to 167 F.

Local code is going to vary, so I can't help there. I would recommend consulting with an electrician also.

wjcarpenter, 2022-08-22:
_That_ is a great answer!

TonyBrobston, 2022-08-23:
Cool. I'm going to close this one. If you have more questions feel free to reopen or create a new issue.

closed 2022-08-23 by TonyBrobston

patricker, 2022-08-26:
@TonyBrobston  Could you elaborate how you got the wire through the vents? I saw a video some time ago where they blocked off all but one floor vent, and then used a a light weight line and some kind of air catch (maybe it was a plastic grocery bag?) to pull the line, and then use that to pull the final cable.

How did you do it?

Nuetron06, 2022-08-26:
I have a mobile home and I'm going to run a remote controlled car through the duct. It's all one level, rectangular and straight along the middle of the floor. Reminds me of a headcrab in halflife lol

wjcarpenter, 2022-08-26:
@Nuetron06 I know that's actually practical, but the whole idea of it just made me laugh. Zoom!

patricker, 2022-08-26:
@Nuetron06 my vents are big flexible tubes, and my house is two stories.

TonyBrobston, 2022-08-26:
I'll post a detailed explanation on a little bit. The cliff notes are:
1. Grocery Bag as a parachute of sorts
2. Jet Line to initial get something through, to then use to pull your wire
3. A hole like 8 in. x 8 in. somewhere near the beginning of your main plenum.
4. The ability to manually turn your blower on (Ecobee has this functionality)
5. Cap all vents except the one you're trying to run the wiring for.
6. You may also need a Fish Tape.

I'll elaborate a bit more it a bit.

TonyBrobston, 2022-08-27:
Long explanation, but I don't like to leave things to chance 🤷‍♂️.

patricker, 2022-08-27:
@TonyBrobston Awesome, thanks for the explanation and all the videos. I have a custom HVAC controller already, with remote temperature sensors. For a while I was using Keen vents over Xigbee, but they haven't been reliable, and the batteries die quickly. I've been tinkering with a project like this for a while... so glad to try out your 3D prints and see how it works.

TonyBrobston, 2022-08-27:
@patricker Out of curiosity, what are you running for a custom HVAC controller?

I have a general list of requirements for smart home products when I buy:
1. No batteries
2. Local Control
3. Prefer MQTT, but not required

I wonder if you could modify those vents and add an ESP chip and hardwire power? Have you opened one up?

patricker, 2022-08-27:
I also like local control and MQTT. I have a raspberry Pi with a relay hat. I run a service on the Pi (written in Python) that subscribes to an MQTT topic and then flips relays to turn AC/Heat/Fan on depending on the message. I then use Node-Red to manage the logic between the temperature sensors, which are also MQTT~ish, and sending the HVAC commands.

TonyBrobston, 2022-09-20:
This seems like the most relevant place to post this. I designed and printed an enclosure for the recommended power supply and power injection board. Its almost ready, but needs a few small tweaks.

[image]
[image]

TonyBrobston, 2022-09-20:
@patricker Looking this over a bit more. I was thinking the inner width dimension of the enclosure was a bit tight, but after putting the three screws in, I am pretty happy with it. 

The only other thing I was exploring was the 110v connection between the power supply and the IEC connector (I think that's the right name). I had to bend the fork spade on the power supply down, bend stranded wire into an "S" shape, and then attach the female spade to the IEC connector. I wouldn't advise doing this without a good crimp; I have a pretty good crimping tool with interchangeable jaws for different types of crimps. I also considered ordering 90 degree female spades, I think that would also help. I tried searching for fork spade 90's or eyelot spade 90's and they look like straight versions of themselves, bent 90; so I think carefully bending a regular version is maybe ok?

[image]
[image]

Designing things with more space seems like an obvious solution, but designing for 3d printing without supports constrained things a bit 🤷‍♂️.

Let me know what your thoughts are.

PInnovation, 2023-05-26:
@TonyBrobston Would you be willing to share the native files for the power supply supports/ housing? I'm looking to do something similar, but with a couple changes to accommodate a larger, external fuse block.

TonyBrobston, 2023-05-29:
@PInnovation Yeah I'm happy to share those. I'll throw it on my list and put a fusion 360 link here within the next few days.

TonyBrobston, 2023-06-06:
@PInnovation Let me know if this works for you: https://a360.co/45KHvFZ