BrobstonCreations/yet-another-smart-vent #8: Missing part?

Nuetron06, 2022-08-21:
Was going through the parts to print some out, but discovered a part that was not in the list of STL files. 
[image]

Is this one no longer included?

TonyBrobston, 2022-08-22:
I added a link to the main readme. There was a link in the electronics assembly, but maybe it's simpler if it's next to the rest of the STL links.

Gotcha, it seems like a lot of people have had this idea and pursued it, but not necessarily to MVP. If you're interested in contributing, I'm certainly looking for help, lol.

Nuetron06, 2022-08-22:
Facepalm... 

I'd love to help if I can. I can't say I'd be available a lot, maybe some on the weekends. 

To get a little more off-topic, some feedback / questions: 

1. The end-cap models seem to be a bit large. I'm new to 3D-printing, but both the stock VoxelMaker and Microsoft's 3D Builder (I haven't tested very many apps) show they are huge. I re-scaled them as best I could. 
2. Would it be better to have an option to print most small parts in one file? I saw you had the arms for the louvers in one file, but others were all separate. 

[image]

TonyBrobston, 2022-08-22:
Be sure to read this before you print the levers. I put those in the same file because of the semi-print-in-place thing I have going on. https://github.com/TonyBrobston/yet-another-smart-vent/blob/master/docs/VENT_PRINTING.md

TonyBrobston, 2022-08-22:
That's odd that the end-cap came in large. I just downloaded on from the 4x10 on printables and pull it into SuperSlic3r (a fork of Prusa Slicer) and it comes in just fine. 
[image]

TonyBrobston, 2022-08-22:
> I'd love to help if I can. I can't say I'd be available a lot, maybe some on the weekends.

No biggie. If something comes to mind let me know. Otherwise I'll likely plug away at what I have listed in the readme. Progress is somewhat slow flying solo and when I take a job here soon, it will probably slow a bit more. Hopefully if this takes off there are some devs interested in contributing. The hope is to also keep things generic enough that Home Assistant can do a lot of the scheduling, automations, etc. I'm also hoping to experiment with moving this repo to ESPHome and move mqtt-hvac-vent-control to a Home Assistant integration (so Python rather than TypeScript), but I need to do a proof of concept to make sure it's viable.

Nuetron06, 2022-08-22:
RE: pausing; That's a good point. I've been trying to find out what app I can use to add the pause in, and it looks like Cura does, but it doesn't natively support my printer (Voxelab Aquila X2). VoxelMaker (came with the printer) doesn't seem to have that capability. 

Regarding the part size... I don't know. At first I thought it was VoxelMaker when I brought it in, but then it did it in Builder too. 
I just re-downloaded the files to get for comparison, but now it's not doing it anymore. The file I had was the #2 in the list of 8 you had, but it's not the same anymore. Go figure.

Nuetron06, 2022-08-22:
There's more I'd like to review, and more questions. Is this issue the best place, if not, where would you prefer to correspond?

TonyBrobston, 2022-08-22:
Here is fine. Ask away. Also feedback is much appreciated. I'm doing the best I can, but I used to collaborating with a full team, lol.

TonyBrobston, 2022-08-22:
> I just replaced my nest thermostat with an esp32 relay board and have it centrally controlled via ESPHome

Out of curiosity, do you have more information on this? Is this something you made from scratch or another project? It would certainly be nice to have more control over the thermostat and the devices it controls.

Nuetron06, 2022-08-22:
I replaced my Nest thermostat with this [Lilygo ESP32 Relay Board](https://www.aliexpress.com/item/3256802978746569.html)
Right now it's just a dummy board controlled through HA & ESPHome. I'll update my HA community thread posted above with my current code. I'm on lunch right now so I have some time.

Nuetron06, 2022-08-22:
Thread updated:
 [Smart Vent Zoning control](https://community.home-assistant.io/t/hvac-smart-vent-zoning-control-through-homeassistant-no-internet-required-but-recommended/440051/2)
 
I'm planning on using the thermostats to open the vents if either cooling or heating is requested. Otherwise, if the temp in the room is in the "sweet spot," the vent will be closed (maybe opened when 1 degree away from cooling/heating threshold?)

Nuetron06, 2022-08-22:
> Out of curiosity, do you have more information on this? Is this something you made from scratch or another project? It would certainly be nice to have more control over the thermostat and the [...]

That's why I got rid of my Nest, as well as getting away from local->cloud->local control of the thermostat, since Suddenlink doesn't notify its customers when it's doing maintenance on the lines...

TonyBrobston, 2022-08-22:
I'll give that update a read in a bit.

It's also worth mentioning that if you use an Ecobee Thermostat with the HomeKit Controller Integration through Home Assistant, you gain local control. However, it's just to set temp, read temp, change mode (heat, cool, off) and fan mode (auto, on). This is essentially enough, but maybe there are other things worth controlling, like for example a blower may have many speeds and that may be worth adjusting.

TonyBrobston, 2022-08-23:
I'm going to close this issue since we addressed the missing part. 

Feel free to continue to comment here or open a new issue if I miss your question. I've been looking through these issues when I'm not heads down into some improvement, lol.

closed 2022-08-23 by TonyBrobston