BrobstonCreations/yet-another-smart-vent #9: Request: 12x12 vent design

dRuEFFECT, 2022-08-22:
I have 6x12 flair vents in my basement and love them, but the rest of my house has 12x12 ceiling vents for central air, about 13 of them. I've been dying for a smart vent solution to fit 12x12. I don't have a 3d printer myself, but I would send them out to get printed so I can give this a try if there's a possibility of getting 12x12 designs up. 

Let me know if there's anything I can help with or provide to make this happen.

TonyBrobston, 2022-08-22:
Putting some more thought to this; if anything we could potentially design for using two servo motors. At that point I think it's likely that those two servos could turn all this mass. It's entirely possible that one servo will do it.

If you happen to take the dimensions, could you also snap a few pictures and upload them? I haven't heard of vents being that large before, so just curious to see the opening.

jpreston84, 2022-08-22:
I'll second this. Other designs are desirable for this to be a viable product -- I've got a 12x12 (approx) in my main room, and 6x8 (I think) in other rooms (this is a house built in the late 1970s).

TonyBrobston, 2022-08-23:
I agree. I'd prefer to make almost every option, however there are a few I'm probably not going to design for... like this one:
[image]

6x8 should be an easy one.

The only downside to all this variation is if changes need made. I do have a lot of the shared models linked, but inherently there are some things that are impossible to share. It may be possible to use parametric design (parameter based values) to help with this, but I'm not quite sure how to have a model as a sort of template and then create multiple models with different parameter values off that. If that's a thing, that would actually work well.

TonyBrobston, 2022-08-23:
Because the 6x8 should be pretty easy, I'll go ahead and commit to that. At some point I'll experiment with 12x12, but it's going to be a bit, at least a week, possibly a few weeks.

TonyBrobston, 2022-08-23:
Here is the 6x8. I have not yet had a chance to test this; at this point I need more than one printer 🤷‍♂️ . Let me know if you do print this and if anything is out of place. If it is good to go would someone mind uploading some pictures here and I'll attach it on printables?

https://www.printables.com/model/264778-yet-another-smart-vent-6x8

TonyBrobston, 2022-08-24:
I added the 12x12 to the readme, it will be awhile before I experiment with it; I need to move on from vent variations for a few days and focus on some other things.

I'm going to go ahead and close this, feel free to reopen or create a new issue if you have more questions.

closed 2022-08-24 by TonyBrobston