BrobstonCreations/yet-another-smart-vent #10: Minor Request: Single Piece Vent Top for 4x10?

joshelson, 2022-08-26:
Just putting this request out there for those of us that have larger format printers and would love the simplicity - and in the case of the very anal retentive among us, the slight aesthetic improvement - of a single-piece vent top.

Any chance of getting a 4x10 as a single printable STL? Would be happy to do the work too, but I believe that part of this project is not going to be open-sourced and 4x10 is by far the most common size for newer homes in the US, at least.

Thanks so much for the immense amount of work you've put into this! Really amazing work.

TonyBrobston, 2022-08-26:
Yep I can do that. Give me a day or so and it should be up.

Thanks for checking the project out. Let me know if you have any other feedback or questions.

TonyBrobston, 2022-08-26:
This was actually the original game plan. Certainly people could machine their own. Someday I may offer to sell these, but I need a CNC that can make these faster than my cheapo CNC does.

[image]
[image]

TonyBrobston, 2022-09-01:
@joshelson I got the 4x10 solid top uploaded to Printables. Sorry it took me a bit longer to get to it than I had said before. At some point in the future I'll likely upload all the vent tops as a one-piece option.

TonyBrobston, 2022-09-02:
I'm going to go ahead and close this issue.

Let me know if there's anything else I can do for you. Feel free to reopen this issue or a new one.

closed 2022-09-02 by TonyBrobston