BrobstonCreations/yet-another-smart-vent #11: step files

Conundrum, 2022-08-27:
Any chance you would share the step files here, I have some very odd sized vents, would love to contribute since I'll need to design other sizes anyway.

TonyBrobston, 2022-08-27:
I'm currently leaning away from releasing this part of the project. This may change in the future.

This is my general explanation as to why: https://github.com/TonyBrobston/yet-another-smart-vent/issues/6#issuecomment-1221664860

Could we start with some pictures/dimensions of your vents?

I'm more than likely fine with putting some other sizes together; hopefully what you have is odd... but standard somewhere.

hubertpi, 2022-08-28:
I was hoping for the same thing as I wanted to try and install the mechanical part into these plastic registers off [Amazon](https://www.amazon.com/dp/B007EMTX2I?psc=1&ref=ppx_yo2ov_dt_b_product_details)

They're very cheap but still look better than a 3D printed register, come in many colors and the backside is a simple sliding louver that is easily removed creating a simple case.
(although I'm debating on attempting to simply motorize the louver that's there)

I completely understand your reasoning for keeping the design files private but if you have a spare moment I would greatly appreciate a variant that fits the registers off Amazon

The inner case dimensions are ~91mm x 242mm x 28mm

I know this isn't a "standard" size but reducing one of the largest pieces to print and eliminating the one that requires the wife approval would probably ease the entry level for many.

Conundrum, 2022-08-28:
Yeah that works for me, I'll see if I can get pictures and measurements this week

TonyBrobston, 2022-09-19:
> Yeah that works for me, I'll see if I can get pictures and measurements this week

@Conundrum Let me know if you happen to get those dimensions. If you do, feel free to open a new issue with those and I'll eventually carve out some time to investigate a solution.

closed 2022-09-19 by TonyBrobston

TonyBrobston, 2022-09-19:
> I was hoping for the same thing as I wanted to try and install the mechanical part into these plastic registers off [...]

That's certainly an interesting thought. I think that approach is probably viable. The way the electronics enclosure and top mount to the vent shell would need to change, along with the length of the louvers. I would tend to think you could print a jig to drill holes in the end of the vent shell for the louvers to insert into. If I was in your position, I would probably download the STL files for a similar size, measure the different components and draw up your own variation.

> I know this isn't a "standard" size but reducing one of the largest pieces to print and eliminating the one that requires the wife approval would probably ease the entry level for many.

On that note, at some point I'd like to offer CNC wood tops. This was my original plan, but there are a handful of challenges that come along with that.
[image]
[image]