BrobstonCreations/yet-another-smart-vent #13: Use ESP32 instead of ESP8266?

Nuetron06, 2022-09-02:
I'm working on a project that will be using BLE, and because the 8266 doesn't support BLE, I'll have to upgrade to a 32. 
Using the below module, what will this add in terms of space required?
https://www.amazon.com/Teyleten-Robot-ESP-WROOM-32-Bluetooth-Development/dp/B09C5RDZ8G
Dimensions are 40mm*34mm.
Existing D1 mini module is 34.2mm*25.6mm

TonyBrobston, 2022-09-02:
One way or another, it's definitely possible to make an ESP32 work. I had considered it at one point, however at the time it didn't make a lot of sense. The ESP32 is bigger, slightly more expensive, and I didn't have plans for consuming the Bluetooth. I do however have interest in figuring out how to run the vent's code/firmware and ESPresence; if that is doable I'd gladly design a variation for both ESP8266 and ESP32 D1 Mini.
This probably doesn't add anything to this conversation, but I am currently moving this project to ESPHome. I have a configuration in a branch that is currently working.

[image]
[image]
[image]

Nuetron06, 2022-09-03:
I did see that one. Looking forward to being able to use it. I'm planning on using the thermostat integration on the esp itself to manage the vent being open/closed. When it's outside the comfort range, it will open the vent, at the same time as calling for heat/cool, and using a serial communication through the in-duct wiring as a backup for the wifi being down. This way a distributed cooling system can operate and still centrally controlled. I'm planning on using another esp to replace the physical thermostat and provide a physical interface to manage the system in the event wifi / HA is down.Sent from my T-Mobile 5G Device

TonyBrobston, 2022-09-03:
I gave parts of this a read: https://github.com/ESPresense/ESPresense/issues/73

Sounds like the marrying of ESPresence and ESPHome is pretty unlikely.

Nuetron06, 2022-09-04:
So, does that rule out expanding the electronics housing?
I'm going to give ESPresence a read.

Nuetron06, 2022-09-04:
Interesting. Have you given this one a read?
https://www.cnx-software.com/2022/08/08/esp-wifi-csi-detects-humans-with-wifi-signals-only-no-sensor-needed/

TonyBrobston, 2022-09-19:
> So, does that rule out expanding the electronics housing? I'm going to give ESPresence a read.

Well, I wouldn't say "ruled out". If we can come up with a scenario where moving to an ESP32 would add benefit and and prove it out, I would be willing to experiment with an ESP32 D1 Mini. Based on my research, whatever benefit we get from the ESP32 would have to work through ESPHome, because it appears that it's a "one or the other" situation. Where ESPresense and ESPHome cannot both be run on one device at the same time.

> Interesting. Have you given this one a read? https://www.cnx-software.com/2022/08/08/esp-wifi-csi-detects-humans-with-wifi-signals-only-no-sensor-needed/

That's interesting. I imagine it wouldn't work as well as ESPresence, but that's just a guess. It also looks like a "one or the other" situation though.

At some point I would like to build a device that reads the rooms temperature, allows you to set the room's target temperature, and displays both of those. I wouldn't be surprised if ESPresence could do this on top of it's presence functionality (again, just a guess). If that is true, we would have arrived at a viable solution.

Nuetron06, 2022-09-19:
> At some point I would like to build a device that reads the room's temperature, allows you to set the room's target temperature, and displays both of those.

This is currently what I'm working on. Right now I have just a tablet for the whole house, with a separate virtual thermostat for each room, using one esp32 installed inside the central hvac unit (the metal panels act as a weak faraday cage, need an external antenna). I haven't gotten the vents set up yet, but it does pretty well so far. I'm using Govee H5075's & H5100 for the rooms. 

I'd like to link each of the vents to the controller either via hardwire or wifi, and a unit with physical interface where the thermostat on the wall used to be. ESPHome / MQTT will be only for linking to HA for advanced management, but the entire system can work independent of HA. That's the reason for the BLE on the vents, so they can read the room's temp from the Govee and act accordingly, as well as forward the temp data to the central controller.

TonyBrobston, 2022-10-27:
> This is currently what I'm working on. Right now I have just a tablet for the whole house, with a separate virtual thermostat for each room, using one esp32 installed inside the central hvac unit [...]

Right, so this is the same thing, except there's no physical device that allows you to set the room's target temperature? I went down a similar route with the virtual thermostats (https://github.com/BrobstonCreations/mqtt-hvac-vent-control). Virtual thermostats are ok, but additionally a physical thermostat would be ideal.

> I'd like to link each of the vents to the controller either via hardwire or wifi, and a unit with physical interface where the thermostat on the wall used to be. ESPHome / MQTT will be only for [...]

At some point I'd like to experiment with a hardwired vent. I'm not sure if [this is the exact right one](https://www.mouser.com/ProductDetail/Olimex-Ltd/ESP32-POE-ISO?qs=unwgFEO1A6vVfCRbLeeCIw%3D%3D&mgh=1&gclid=CjwKCAjw2OiaBhBSEiwAh2ZSP6-aY7QhAAUetFx9FwnvVmFLyasD0HwlhYM-mvkPiXIFMEl-aN-2MRoCmtcQAvD_BwE), but there are some ESP32 POE options. They are expensive in comparison to the ESP8266 D1 Mini's... but I suppose that's just how it goes 🤷‍♂️ . It will be an odd one to design an enclosure for, but it should be possible with enough time and effort. Though, I think it's a hard one to justify as the amount of work is slightly greater (especially when there's a wifi option) and the number of people interested is likely lower. But, that doesn't mean I won't eventually do it.

> I plan to have a virtual thermostat on the vents too, only controlling the open/closed state of the vent, so that the vent will open before the HVAC kicks in and close before it turns off, e,g;
> Room TStat temp range (heating/cooling): 68-72°F
> Outside this range, HVAC will run
> Room Vent: 69-71°F
> Outside this range, vent will be open

It sounds like our use-case is slightly different. Are you saying acceptable temperatures for rooms are a range rather than a target number? If so, will all rooms be the same or can they be changed?

Nuetron06, 2022-10-28:
I started building a spreadsheet to help explain, but it's not complete. 
https://docs.google.com/spreadsheets/d/1cdWEg5NlcUvEFNqmaFtIvSUsdUBis61yegOo_d3Ahfw

Nuetron06, 2022-10-31:
I'm still thinking it through... I think I need to write my own climate controller integration. The thermostats we have are not designed for multi-zone conditioning.

Currently, a simple thermostat takes action (heat or cool) when the threshold temperature is crossed, and stops the action after the threshold is re-crossed. Add smart vents, thermometer and a separate thermostat for each room, and you have zoning using a directed-air system that will only heat or cool specific rooms.  
Now, to add on top of that... if you have a "comfortable temperature" range, you can use the different rooms to your advantage, as capacitors to store temperature instead of just objects to be managed. 

Scenario 1:
- Room A is above its comfort range, and any other room is below its comfort range: Turn on fan to balance the two, opening appropriate vents. 

Scenario 2 only activates if # 1 has been tested:
- If one is still getting too far out of bounds, activate heat or cool as needed and disconnect air to the one that is still within bounds, since some balancing has been done already.

I already have a rudimentary setup with # 1 using a 2-thermostat configuration and multiple thermometers... 
The thermostat is a dual-point thermostat in heat-cool (average house temp across 3 devices), and the second is a fan-controlling single point, which keeps the rooms at an even temperature (I had some issues with temperatures getting converted between °C and °F, so I added 10°C = 50°F)

[image]
[image]

Here are the code snippets for the thermostats & sensors:
`
sensor:
  - platform: template
    id: house_temp_average    
    name: "House Temperature Average"
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    lambda: |-
      return (id(mrbr_temp).state + id(lvrm_temp).state + id(frbr_temp).state)/3;
    update_interval: 5s    
  - platform: template
    id: house_temp_difference    
    name: "House Temperature Difference"
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    lambda: |-
      return std::max({id(mrbr_temp).state, id(lvrm_temp).state, id(frbr_temp).state}) - std::min({id(mrbr_temp).state, id(lvrm_temp).state, id(frbr_temp).state}) + 10;
    update_interval: 5s    

climate:
  - platform: thermostat
    name: "Whole House"
    sensor: house_temp_average
    min_cooling_off_time: 300s
    min_cooling_run_time: 300s
    min_heating_off_time: 300s
    min_heating_run_time: 300s
    min_idle_time: 300s
    cool_action:
      - switch.turn_on: esp_hvac_cooling_house
    heat_action:
      - switch.turn_on: esp_hvac_heating_house
    idle_action:
      - switch.turn_off: esp_hvac_cooling_house
      - switch.turn_off: esp_hvac_heating_house
    preset:
      - name: Away
        default_target_temperature_low: 66 °F
        default_target_temperature_high: 78 °F

  - platform: thermostat
    name: "House Fan"
    sensor: house_temp_difference
    min_cooling_off_time: 300s
    min_cooling_run_time: 300s
    min_idle_time: 300s
    cool_action:
      - switch.turn_on: esp_hvac_fan_house
    idle_action:
      - switch.turn_off: esp_hvac_fan_house

`

Nuetron06, 2022-10-31:
After looking at this a bit more, I think I need to design a "quad point thermostat" that has the comfort set range, the upper fan range, lower fan range (based on if there is available temperature differential in another room) and heat / cool points. 
Something like this:
[image]

Thoughts?

TonyBrobston, 2022-11-05:
At some point I'll dive into this some more and give you my thoughts.

At first glance, I feel like we should write something together, as that could be a good thing. I did have a guy named Vadim reach out via email about a system [he has written](https://github.com/home-climate-control/dz/). Though my initial thought is that I would prefer a Home Assistant Integration.

I think it's important that whatever controls everything is a Home Assistant Integration, if we are interested in others using that control system. I do realize you don't want to have your system be dependent on Home Assistant, so maybe that's a sticking point. I don't know enough about integrations, but I assume once a device is integrated that it's an "entity" and I'm assuming it's possible to work with "entities", which would make the system device agnostic, which seems like it would lower the complexity of the integration while increasing the number of devices that would work with this integration. I'm mainly thinking different thermostats and temperature sensors, but also other smart vents if people figure out how to integrate them with Home Assistant.

I think this is some higher level stuff that interest me, before getting into exactly which features the system would have. But then again, I don't know if you're interested in mainly building for yourself or keeping others in mind too? or if you're interested in collaborating on some level? I have a lot going on at the moment, but writing a system like this is on my long list.

Nuetron06, 2022-11-05:
> At first glance, I feel like we should write something together, as that could be a good thing. I did have a guy named Vadim reach out via email about a system [he has [...]

I'm totally open to collaborating on a project like that. I like HA (it's got quite a range of flexibility) and want to stick with it. 

> I think it's important that whatever controls everything is a Home Assistant Integration, if we are interested in others using that control system. 

Agreed. I'm a tinker-with-it-until-you-get-it-right kind of guy, but I've learned recently that a good UI is really important. Wives tend to teach you that. Make it simple and they love it.

> I do realize you don't want to have your system be dependent on Home Assistant, so maybe that's a sticking point.

Only in so far as to HA going down and losing control over the system. I've had a few things happen in the last couple months of running my own control board in the air handler that it's important to have the ability to control the HVAC independently from HA (try having your HA host lose its static IP during a reboot and not get a new one from DHCP while you're away from work, and it's the only way to get temp data to the HVAC to make decisions on when to heat/cool. There are also heat-sensitive pets in the house that could have seizures over 85F).

I also like the idea of having a physical "base" master thermostat interface that can be used to flip between the different rooms' settings, typically installed where the old thermostat was. This could be expanded to "satellite" thermostats in each room.

> I don't know enough about integrations, but I assume once a device is integrated that it's an "entity" and I'm assuming it's possible to work with "entities", which would make the system device [...]

I think increasing the range of interoperability is a great idea. Also, building an integration that would give you a GUI to select your sensors/devices etc and having each vent built using the ESPHome API will give greater flexibility, in so far as to people using the HA core version. I'd prefer to move away from the API only because it makes the ESP reboot if no connection has been made in 15 minutes, but that's just me. 

> I think this is some higher level stuff that interests me, before getting into exactly which features the system would have. But then again, I don't know if you're interested in mainly building for yourself or keeping others in mind too? or if you're interested in collaborating on some level? 

I'm open to collaborating and contributing. My original concept was to build for myself, but from the onset, my wife has wanted me to take it to the next step and take on what you have been doing, and possibly make a business of it. However, I don't have a lot of the know-how and would have to take time from other duties at the moment.
The other aspect is that I don't have the computer modeling skills. I can visualize a concept and see all the parts working together, but getting that translated to a visual on a computer is not my strong suit. 

That being said, I really get a thrill from building something that other people can use. Maybe we can partner up on some things? There's a lot of knowledge and wisdom you have that I don't, but I'm a fast learner.

>I have a lot going on at the moment, but writing a system like this is on my long list.

Me too. 

> I had also [written this](https://github.com/BrobstonCreations/mqtt-hvac-vent-control) and would consider it more of a prototype. I quickly realized most people don't seem to know docker or be interested in running a system like this with docker.

I like this idea, and I've been delving into MQTT recently too. 
Yeah, Docker is a bit of an advanced setup that most people aren't going to want to touch. 

>This is one of the other reasons to write a Home Assistant Integration. It would also be ideal to boil the setup process down to be as simplistic to setup as possible; maybe there would be a standard configuration by default and some level of customization built on top of that. I would hope that if a user already has a thermostat, temperature sensors, and vents integrated into Home Assistant and assigned to rooms, that the Integration could detect those and be up and running as soon as they install the integration. Then, they could optionally turn less common features on through some configuration panel or something like that...

The majority of people that are going to gravitate towards this ventilation system will want it to "just work" without having to get into the specifics of how. An integration that has a gui for selecting your inputs and outputs always goes more smoothly than having to edit code.

 Let me know if you want to move forward with collaboration. I've got a lot to learn but I pick it up fast.

TonyBrobston, 2022-12-07:
Whelp, it's been a long month; just getting back to replying now.

> I'm totally open to collaborating on a project like that. I like HA (it's got quite a range of flexibility) and want to stick with it.

Cool 😄 .
 
> Agreed. I'm a tinker-with-it-until-you-get-it-right kind of guy, but I've learned recently that a good UI is really important. Wives tend to teach you that. Make it simple and they love it.

I can definitely see that. Though I think the best automation is the one where it just knows what you want without getting involved, but that's not always feasible; or maybe not yet?

> Only in so far as to HA going down and losing control over the system. I've had a few things happen in the last couple months of running my own control board in the air handler that it's important [...]

Yeah, that's a real scary reality. I'm current on the boat where my thermostat would do the right things even if Home Assistant goes down; I suppose that's one upside of using a commercial thermostat. That's not to say that a DIY one can't be hardened to that point though.

> I also like the idea of having a physical "base" master thermostat interface that can be used to flip between the different rooms' settings, typically installed where the old thermostat was. This could be expanded to "satellite" thermostats in each room.

I could see this being a tablet or something similar that has access to a "whole home" interface. I previously had something like that, but am not currently running the other system I wrote. This also included mqtt hvac thermostat components for each room, where you could set desired temperature for rooms.
[image]
 
> I think increasing the range of interoperability is a great idea. Also, building an integration that would give you a GUI to select your sensors/devices etc and having each vent built using the [...]

I don't think I've experienced this reboot piece. Seems like another way around that would be for ESPHome to add a configuration value to adjust what that does. 

> I'm open to collaborating and contributing. My original concept was to build for myself, but from the onset, my wife has wanted me to take it to the next step and take on what you have been doing, [...]

That's good. I think it's important to put these things out into the world for the concepts to progress; isolation has a lot of limiting factors. It's definitely a weird endeavor; I think there are a lot of opportunities in this space, especially when things like this take off: https://everythingsmarthome.co.uk/the-everything-presence-one-now-available/. I think it's worth pursuing if you have the time, the capital, and the willingness to throw yourself headfirst into the unknown. Life seems to be one big learning concept. But I get where you're coming from, I'd say marketing and being the face of something isn't my strong suit. Engineering however, is an obsession and seems to be endless, which means I can continue obsessing, forever; lol.

> That being said, I really get a thrill from building something that other people can use. Maybe we can partner up on some things? There's a lot of knowledge and wisdom you have that I don't, but I'm a fast learner.

I do too. I think a good place to start is this Home Assistant Integration. I'm sure you know things I don't too.

> I like this idea, and I've been delving into MQTT recently too. Yeah, Docker is a bit of an advanced setup that most people aren't going to want to touch.
>
> The majority of people that are going to gravitate towards this ventilation system will want it to "just work" without having to get into the specifics of how. An integration that has a gui for selecting your inputs and outputs always goes more smoothly than having to edit code.

Agreed. I thinks it's entirely realistic to make it super simple to get up and running, but requires more work; but that's just how that works. The easier it is for the end user, the more time the development team spent on it.

> Let me know if you want to move forward with more in-depth collaboration. I've got a lot to learn but I pick it up fast. I'm also full of ideas to fill needs, but certain ones need to be prioritized over others or you never get anything done.

Yeah. When I am able, I'll create a GitHub repository and begin going through this documentation: https://developers.home-assistant.io/docs/creating_component_index/. 

The initial goal would be to create a "hello world" integration that can be installed into Home Assistant and do something as simple as display "hello world" somewhere. Next, I would think it would be important to understand how an Integration sends and receives data from components within Home Assistant and if we can tell what components exist within Home Assistant "areas". After that, we would write basic functionality to read actual room temperatures and target room temperatures, adjust the main thermostat's temperature, and adjust vent positions. I think the biggest unknown here is how to create virtual thermostats; this may spin off into another device, something like this: https://github.com/ioios-io/demos/tree/main/Home%20Assistant%20with%20ESPHome/3.%20Climate%20Control. In other words, we would have a physical device for each room that can read actual temperature, set desired temperature, and display both. If this was another ESPHome device, it would then become a component within Home Assistant.

Again, I'm not sure when I'll get to this. I will likely start a new job shortly after the new year, so that also doesn't help matters. Unfortunately something has to pay the bills 🤷‍♂️  and none of this is doing that, at least not yet.

Nuetron06, 2023-03-16:
https://www.reddit.com/r/esp32/comments/xjcof5/esp32_in_true_d1_mini_form_factor/

Nuetron06, 2023-03-16:
Even better!
https://www.cnx-software.com/2022/03/18/lolin-c3-mini-esp32-c3-board-compatible-wemos-d1-shields/

TonyBrobston, 2023-03-17:
That's kinda crazy.