BrobstonCreations/yet-another-smart-vent #15: Split 4x10 vent shell to work with Creality CR6-se

bretterer, 2022-09-03:
This is such a great project and I love it, but I am unable to print this file due to my bed size restriction.  

You currently split out the *x12 files in half to be able to print on smaller machines.  I am wondering if you would be kind enough to provide a split version on the *x10 files as well.

TonyBrobston, 2022-09-03:
@bretterer Thanks for reaching out.

From a quick google search, are the bed dimensions 235 x 235 x 250mm?

bretterer, 2022-09-03:
@TonyBrobston that's correct.  It seems that the frame model is just too big though.  
 I typically run max of 225*225*235 to give some breathing room

Nuetron06, 2022-09-03:
I've got an Aquila X2, it's about the same size. However, I've had to split it myself (actually, my wife did it for me) because mine will support the size, but only when on-end, except my printer has difficulty with printing supports.

TonyBrobston, 2022-09-03:
I print on a Prusa i3 mk3s+. The build plate is 250 x 210 x 210. The 4x10 vent shell just barely fits. The 4x10 was the first vent I designed and the shell is drawn this way to work around the size constraint.

[image]

I'm just trying to make sure it isn't possible before I make another shell with a snap joint, as they take an hour or so to draw. I think the amount of variations that I'm creating is going to make future modifications really hard to maintain, that's another reason I ask.

bretterer, 2022-09-04:
This is what it looks like on the creality cr6-se which is 235x235

[image]

TonyBrobston, 2022-09-04:
Oh, I'm a derp. So, the 250mm is the Z?🤦‍♂️

TonyBrobston, 2022-09-04:
I'll plan on making a split shell with a snap joint for the 4x10 and 6x10. But I'll also plan to keep the regular shell too. It will likely be a few days until I get to that, but probably less than a week.

bob5731, 2022-09-04:
I would also like a split shell for the 4x12 I have an ender 3 pro.

TonyBrobston, 2022-09-04:
@bob5731 I uploaded a 4x12 split shell a week or so ago. https://www.printables.com/model/259924-yet-another-smart-vent-4x12/files

bob5731, 2022-09-04:
I also think I need a 6x12 what. Does it cover on the 4x12 and the 6x12. Will not fit in my prayer if I remember correctly.

TonyBrobston, 2022-09-19:
> I also think I need a 6x12 what. Does it cover on the 4x12 and the 6x12. Will not fit in my prayer if I remember correctly. I was also working on a similar project. Where I take it off the shelf discontinued timer vent with a motor controller and rig it up to an Arduino to run through SmartThings.

Your message is hard to follow. It looks like you used voice-to-text and it went horribly wrong.

There is a 6x12 vent: https://www.printables.com/model/259622-yet-another-smart-vent-6x12

> Your project may be a lifesaver since I live in a two-story condo in poor temperature control. If there's any way in the future you could implement this to smart things I would appreciate it.

I am not familiar with Smart Things. I would tend to think that there is probably a way to use MQTT with Smart Things; I would start by researching that. If that's possible, then you'll need to enable MQTT in the ESPHome configuration when flashing the device.

TonyBrobston, 2022-09-19:
@bretterer I am starting the 4x10 vent shell split now.

TonyBrobston, 2022-09-19:
@bretterer Turns out I was able to copy the 4x12 shell and adjust the length dimensions, so that went much faster than anticipated.

Those split shell should be out there now: https://www.printables.com/model/259241-yet-another-smart-vent-4x10

Let me know if you have any other questions or requests. Thanks!

closed 2022-09-19 by TonyBrobston

hidden0, 2022-10-11:
For what it's worth the 2x12 split shell functions amazingly.