BrobstonCreations/yet-another-smart-vent #18: ESPHome yaml file issue (vent does not open)

xeryax, 2022-09-08:
I've messed with the yaml file for ESPhome a few times, and my vent doesnt seem to open...
the open value of the pot is .039, the closed value is .433 (when i move the servo manually) but when i turn on the device, it doesnt calibrate a value for open.
When I try and move the louvers, the device always seems to go the same direction (towards "1" on the pot) it doesnt move the other way no matter which button i press on the dashboard, or whether i move the slider around on HA.

I will poke in the yaml a bit more tomorrow and see if i can create a PR if i find the issue, but its time for bed :)

TonyBrobston, 2022-09-08:
Can you provide a bit more information?

Are you using the Batan B2122? If so, did you align the wiper arm to 60 degrees of the 120 degrees of travel when installing it?
When you are testing, is the vent assembled?

I could guess at the problem and potential solutions, but let's start with those questions and we'll get things working.

Thanks for reaching out.

xeryax, 2022-09-08:
> I could guess at the problem and potential solutions, but let's start with those questions and we'll get things working.
You already did guess at the problem- it wasnt at 60Degrees when i switched from "testing/unassembled" to assembled.

Sorry for the  trouble, but I definitely will leave this here so everyone can see this as a "FAQ". :)

Thank you so much

closed 2022-09-08 by xeryax

TonyBrobston, 2022-09-08:
No problem! I'd prefer people reach out either way, it helps me see the heartbeat of the project.

It's hard to tell how many people are using this and whether they're having problems or not. I'm here to help!

I'll put a quick ESPHome configuration together that turns to 60 degrees (technically servo 0 position in ESPHome). I was planning to do this already, just hadn't gotten to it.