BrobstonCreations/yet-another-smart-vent #23: Powering questions

hendrix04, 2022-10-07:
Thanks for this project!

I am curious as to why you suggest the 24v/15A as a power supply option when the ESPs run on 5v.

I also am curious as to why you decided to use the power shield instead of just powering the device off of USB.

TonyBrobston, 2022-10-07:
Thanks for reaching out.

The short answer is voltage drop when using a central power supply.

Here is a more detailed answer I gave to the same question: https://github.com/BrobstonCreations/yet-another-smart-vent/issues/2#issuecomment-1221040638

hendrix04, 2022-10-07:
Awesome, thanks so much! Apologies for opening a duplicate issue. I glanced through the closed tickets but didn't see one that I thought was relevant.

closed 2022-10-07 by hendrix04

TonyBrobston, 2022-10-07:
No problem!

Let me know if you have any other questions.