BrobstonCreations/yet-another-smart-vent #24: Parametric model

jlk, 2022-10-09:
Hey Tony - thanks for all the work on this! For whatever reason, it's more attractive to me than the commercial options...

I've been tinkering with the STLs for a week or so - turns out the 4x10 shell is about a cm too big for my vents, so started bringing pieces into Fusion to scale. Then realized it made sense to have all the components in Fusion so I could verify the model lengths before printing.

My first attempts were just to scale the lengths, but I'm pondering redoing some of the components with dimensional parameters. Any interest in a PR for a f3d file if I make progress here?

Also, I suspect seeing the components together in a model would help people figuring out how to assemble the vent...

[image]

TonyBrobston, 2022-10-12:
I've been meaning to respond to this. I'll give a thoughtful response in the next day or two.

Thanks for reaching out.

TonyBrobston, 2022-10-13:
> Hey Tony - thanks for all the work on this! For whatever reason, it's more attractive to me than the commercial options...

Thanks :). I feel the same way and mainly created this because I wanted a WiFi (or maybe someday ethernet) option that keeps communication within my network.

> I've been tinkering with the STLs for a week or so - turns out the 4x10 shell is about a cm too big for my vents, so started bringing pieces into Fusion to scale. Then realized it made sense to have all the components in Fusion so I could verify the model lengths before printing.

Out of curiosity, do you know the dimensions of the duct opening? I tried to account for the standard sizes and sizes by request, however I'm sure there are all sorts of unexpected things out in the wild.

> My first attempts were just to scale the lengths, but I'm pondering redoing some of the components with dimensional parameters. Any interest in a PR for a f3d file if I make progress here?

Putting out an f3d or step file has been something that's racked my brain quite a few times. On one hand I want to give the people what they want, on the other hand I'm afraid that there may be a few people out there with malicious intent. I hope to offer fully assembled vents as a product in the future, while also allowing people to manufacture their own for personal use. Right now I'm thinking I may release an f3d file once I am ready to sell fully assembled vents. It looks like I can upload `.f3d` to Printables and I have a license on there. If I find that people are doing something malicious I'll take that one file down and chalk it up to a lesson learned.

I have used parameters a small bit in Fusion 360; this certainly feels familiar given my programming background. Using a common parameter throughout a design seems to increase ease-of-use for adjusting dimensions across multiple pieces of a component. 

I've wondered for awhile if it's possible to tie the number of copies of a single linked component to a parameter. For example, a 4x10 vent has four louvers and a 6x10 vent has six louvers; I'm wondering if it is possible to have a `number-of-louvers` parameter that would use `4` in the first example and `6` in the second? This does not account for joints between components, so if this is possible, that's another small hurdle to overcome. I bring this up as a stepping-stone to decreasing long-term maintenance and increasing speed-to-market (or at the very least, decreasing the time it takes between an idea forming and completing the work for it). For example, I am working to add a limit switch to more accurately determine the closed position. I currently have 2, 3, 4, and 6 inch wide vents and am looking to support two servo motors. This results in 8 different electronics enclosures and as a result, a maintenance nightmare. The difficulty is that the number of levers, louvers, holes in the linkage, and openings in the electronics enclosure is relative to the vent width. I think if there's a way to solve what I'm describing, I will use parametric design as much as I can for other parts of the design, assuming it adds value.

> Also, I suspect seeing the components together in a model would help people figuring out how to assemble the vent...

I agree. At an absolute minimum, I'll reupload the STL of the vent as an assembly. But I do think an actual f3d or CAD file would be more ideal.

TonyBrobston, 2022-10-13:
As I do some more research. I've come to a potential solution.

Currently I have individual f3d files for each vent size. Within those individual f3d files every component is a linked component except for the vent top and vent shell.

I think Fusion 360 must lack the ability to "Insert into Current Design" the same component multiple times based on a parameter (maybe this is possible, but I couldn't find anything). So, I think I will have to do this manually, which seems mostly fine, because I also need to set up joints between the inserted components.

It does seem that I can make my long louvers parametric and adjust the parameters specific to the vent dimensions, which means all long louvers could be derived from a single component, which should decrease long-term maintenance (some) if I make changes to the long louver.

It also seems that I can using parameters in a pattern, which should allow me to derive all my electronics enclosures (and lids) from one file. This video seems to hit on that concept pretty well: https://www.youtube.com/watch?v=lzJkxo9Kf9o.

I think that accounts for most of the hurdles I felt that I needed to overcome to make all this viable. I would tend to think by the end of this, it would be possible to adjust the width and length parameters and end up with a vent that is slightly undersized or oversized compared to the original f3d file. At some point this concept sort of falls over though; for example if you scaled a 6x12 vent to 12x12, there will likely be issues related to how much mass the servo can move.

Another long-term maintenance issue is related to releasing STL's when changes are made to the f3d file. Ideally there would be some sort of CI/CD process that would help remove that redundant work, but one thing at a time. I did some quick googling, it doesn't appear that Printables has an API for uploading. I did leave some feedback though.

TonyBrobston, 2022-10-13:
After a few hours of work, it has become apparent that "Insert into Current Design" creates a link. It seems that it is not currently possible to update a linked component's parameter. If you right click and choose "Break Link", you can then update the component's parameter. However, this completely defeats the purpose. I think what I'm looking for is a templating concept, it sounds like this will be available with Fusion 360 Configurations when they are finish implementing that as a feature.

For now I suppose I will design things as though templating is possible, then break the link, and update the component's parameter. Then later down the road when this functionality is possible, I'll delete the component and re-"Insert into Current Design", and hopefully update the parameter without breaking the link. 🤦‍♂️ 

Here's an example of the linkage being parametric for the never of louver/linkages. This is just changing the parameter from 6 -> 4 -> 3 -> 2.
[image]
[image]
[image]
[image]

TonyBrobston, 2022-10-14:
Moving on to the enclosure:
[image]
[image]
[image]
[image]

I'm really wishing Fusion 360 supported conditional parameters.

jlk, 2022-10-19:
Sorry for slow response - Printer decided to go haywire on me so that took up my free time for a few days. 🙄

And yeah, I get the malicious intent part. Hopefully there's a balance between lowering maintenance vs dealing with the miscreants. Might not be worth the overhead, but maybe patreon or something similar and offer the f3d for a few bucks? 

Wow - my little brain was just thinking of length/width, but yeah I get that it's way more complex than that. Patterns+parameters should help, but might require reworking things a little. And I'm with ya - f360 frequently seems to be on the edge of being great at some of the more complex things, but you end up having to jump through hoops.

At first I was just trying to use parameters to scale - that seemed to work, then I realized slots to locate the electronics box were no longer in the right place, so I sliced the shell (see line in image above) and was scaling just the right side. 

Lemme know if I can help as you tinker with that stuff. :) 

So on my current vent:
* The insert that goes into the duct is 3.75" x 9.5" x 1.4375"
* The duct opening is 4.125" x 10" - but the issue is the sides are slightly slanted so the two dimensions decrease after about .5"

dgi816, 2023-04-26:
Hello Tony and John
Hey, sorry for building on top of your question John, but I came across this issue as well and was going to issue a pull request and found you asked the same thing I was going to ask

Tony, just as John and perhaps a bunch of other individuals, thanks for all your work.
Have you been able to create the models in Fusion 360 or any other parametric software and would you be willing to share?
I have the same issue, my home is older and the ducts (wall) are bent or not standard so I was going to start designing my own based on your STL files but I am not that good on modeling, so I figured I would ask first.

Thanks and please let me know when if you have a link to share them

GregoryHlavac, 2025-01-18:
Did this ever come around to anything?

Have a voron that can.. Very much just print things probably full-sized all in one for most vent sizes.