BrobstonCreations/yet-another-smart-vent #25: Requests; Invert and More Positions

codefaux, 2022-10-14:
Hi there, beautiful project. Thanks so much for providing it.

I've just built one, using different steppers (and requiring simple modifications to two of your models) and my first issue is that it shows "Open" when the vent is closed, and "Closed" when it's open. Clicking Open closes the vent, clicking Close opens it.
- Any way to allow for a true/false parameter to invert Open/Close direction?

Second, our home has no 'return' ductwork - just a central intake in the hallway. My intent is to use these vents to control airflow more precisely than just on/off. We really don't need granular 1% level controls, just a bit of coare adjustment.
- Would you be willing to add support for "Open/Half/Close" or "Open/Quarter/Half/Closed" or similar? Or even just direct percent/position control...

Third, your README.md states `by only closing vents (that are in a closed state) to something like 80% closed` -- I don't understand how this is possible, given your provided control scheme.
- Is there an adjustment I haven't noticed for Open/Closed position bias? IE, "Open" is 85% position, "Closed" is 15% position?
- If not, would you be willing to add one?

codefaux, 2022-10-14:
Hey, thanks for the super detailed reply. I haven't sat down to do the thing yet, but I'll poke at it with a stick today probably.

A few further inquiries, though you've solved the meat of my issue.

`two servo motors that drive opposite directions` -- This....sounds like a big downgrade to me. One servo must overcome the other's idle torque? Or do both servos move both directions, and only one does work at a given time? I can't really conceptualize how this works or what bonus it provides, but it needs
- more power
- more space
- more wiring
- more parts/money
Help me understand?

`and you add a limit switch`  -- Okay; I am _so_ not trying to be confrontational, but... I don't want to do that. I have a servo, I modified it to add feedback, and your calibration routine works so long as the print is good enough and the servos are strong enough. (Note; I DID need to modify your original STLs to make more space because front-face-loading on my servos made them lock and freeze nearly 100% of the time)  I chose this project because it required only the essentials; power supply, ESP, servo, printed parts. I would've liked to see a relay for servo power (for example, mine stuck during an operation and the servo **literally melted its motor spur gear** because the motor _never turned off_ -- I'd suggest stopping PWM after your configured transition time + ~500ms) but that's _all_. A limit switch makes some degree of sense, I suppose, but I'd prefer an option not to use it.

`tilt control added` -- Fantastic, glad to hear that. I uh...may not have ever operated with ESPHome before. Our home is purely Home Assistant + MQTT + Tasmota. I'll readily admit to having not looked lots, but that's because I have no solid footing with ESPHome and have zero desire to break our in-use residence trying to blunder my way to making this work, lol. I don't need you to make me a walkthrough or anything - if you'd be so kind as to just point at a doc that's relevant, I'll go nuts on my own. I simply don't trust myself to find the best way with the fewest caveats, the first time through. Ideally that's not an iritating ask...

`documentation that is slightly out of date or just confusing` -- Fair go, no worries. I open-source frequently, lol, just wanted to make sure I wasn't being obtuse.

Also, I've attached the two files [yet-another-smart-vent-6x12-modded_files.zip](https://github.com/BrobstonCreations/yet-another-smart-vent/files/9788414/yet-another-smart-vent-6x12-modded_files.zip) which I had to modify to make this work for me with my Amazon 8ea for $14 "SG90 9G" servo set, after adding the single Feedback wire to the servo. I used FreeCAD, imported the STL files, and made direct modifications. The electronics box and lid both needed tiny changes to support my servos. They're the 6x12 design. I'd suggest the same changes on your other designs. Reasoning;
- The screw tab on my servos would not fit in the gap provided, either tab width or thickness (this is likely to happen with many servos)
- The servo body is slightly longer so it runs into the back wall, causing the front gear to press into the lever assembly, causing it to jam with near 100% consistency (same, likely to be common issue w/ cheap servos)
- I cut the tab seats up to allow slide-from-rear installation into already-seated lever assembly (easier install, easier servicing, less frustration) instead of slotting it from above while trying to hold/move the lever assembly in place with it
- Placed holes for servo mounting screws (screwless, the tilt inconsistency caused by the "tab in slot" design resulted in frequent calibration problems) and punched a hole for the back of the servo.
- The new servo installation hole is covered by the vent case when mounted to it by the three screws, so it's not a concern that it's "open"
- Not necessarily important (more of an accidental inclusion) but I knocked the pointies off the Lid piece -- PrusaSlicer produces empty layers below printable layers near the tips and complains about it, known issue for very small vertical elements

- New install sequence w/ my parts; levers onto electronics box, servo in from rear already connected to ESP w/ headers or solder, screw into lower servo tab, install lid, break out thin layer interrupting installation hole if required (won't print easily without it) screw into upper servo tab

Also note, you edited while I was replying so I re-read and I'm mostly sure I caught the changes but if I failed to address something or questioned something you clarified, bear with me lol

TonyBrobston, 2022-10-14:
I'll respond to the other pieces when I have some time.

> `two servo motors that drive opposite directions` -- This....sounds like a big downgrade to me. One servo must overcome the other's idle torque? Or do both servos move both directions, and only one [...]

I now realize I created some confusion. There will only be one servo in the vent. I'm just saying, there are two servo motors that I plan to provide support for, the Batan B2122 and DFRobot DMS-MG90-A. In ESPHome if you tell the motor to go to 0, then go to -.5, on one of those motors, it will turn clockwise; on the other it will turn counterclockwise. Does that make more sense?

codefaux, 2022-10-14:
`I'll respond to the other pieces when I have some time` - No worries, I don't mean to be a bother. I appreciate your attention and willingness to clarify/converse.

`Does that make more sense?` - Absolutely, thank you for the clarification. It was probably clear to everyone but me anyhow, lol. This is my second project using servos sum total (literal RC vehicles aside) so I'm definitely a little behind the curve.

`providing support for both` - Completely justifiable. I wasn't aware there were significantly different implementations of the protocol/behavior, I had assumed "a servo is a servo, just some have feedback wires included" aside from supported input voltage range differences.

Take your time on replies, I'm not a priority.

TonyBrobston, 2022-10-20:
> `and you add a limit switch` -- Okay; I am _so_ not trying to be confrontational, but... I don't want to do that. I have a servo, I modified it to add feedback, and your calibration routine works so [...]

This is something that has weighed on my mind for months. I tried really hard to do it all with just a potentiometer and some out-of-the-box thinking. In the end it really came down to two reasons:
1. ESPHome does not make writing complex logic easy. If you compare these two files: [with only using potentiometer](https://github.com/BrobstonCreations/yet-another-smart-vent/blob/cfd305696bf1ebec8c350c790dee858300c16645/yet-another-smart-vent.yaml) vs. [with using a limit switch](https://github.com/BrobstonCreations/yet-another-smart-vent/blob/master/common/yet-another-smart-vent-base.yaml). What you'll find is, a lot less code, and something that is easier to comprehend. All of this for like 50 cents. Also, all the extras that come with ESPHome far outweighed writing all this as an Arduino project, which is how this project started.
2. When I eventually offer to sell assembled units (to those that can't or don't want to build their own), I need them to be solid and I need them to work 99.9% of the time. The potentiometer approach worked like 95+% of the time. It's possible I could have taken things further and got closer to that 99.9%, however I still wouldn't have address my first point; if anything there would be even more complexity and even harder to comprehend.

> `tilt control added` -- Fantastic, glad to hear that. I uh...may not have ever operated with ESPHome before. Our home is purely Home Assistant + MQTT + Tasmota. I'll readily admit to having not [...]

I picked up ESPHome only a few months ago. It's pretty straight forward if you have some Arduino experience and understand that they're trying to abstract away super technical concepts, so that you can get complex things working faster and easier. Someday, when I offer to sell these vents; integrating a vent into Home Assistant will be as simple as the [this](https://youtu.be/D073FX6kdh8?t=472). For now, you'll need to flash your own ESP8266 D1 Mini. It's actually really friendly and inviting if you ask me, [here's a screen capture I put together of it](https://youtu.be/D073FX6kdh8?t=324).

> Also, I've attached the two files [...]

I spent a lot of time working with knock-off SG90's. I've destroyed 5-10 of them. I've had days where I'm developing for 5-8 hours straight and am asking a $2 servo to turn 90 degrees every 2 minutes for a few hours. It's safe to say, they are _ok_... but I wouldn't recommend them to anyone. Back to my other point, I want a project that is rock solid; so working with anything that I can make fail, doesn't really fit that. I realize that I only recently added support for the DFRobot DMS-MG90-A, but that is a $6.30-6.90 servo, that seems to be good quality. If we can find another cheaper servo that (truly) has metal gears and preferably a presoldered/prewired position sensor wire, I would consider adding support.

> Also note, you edited while I was replying so I re-read and I'm mostly sure I caught the changes but if I failed to address something or questioned something you clarified, bear with me lol

Gotcha. Yeah, don't worry about it. I'm here to help the best I can and with what time I can carve out.

TonyBrobston, 2022-10-20:
It's also worth mentioning, you can continue to use the setup that does not use a limit switch. The first tag/release will always be there. But, if there are bug fixes, you wouldn't get them without some effort. But, there really are a limited number of things you can do to a vent, so there probably isn't much to worry about 🤔.

codefaux, 2022-10-26:
Apologies this took so long, took a little time away. Looking back over what I'm about to send, the big end of it appears to be that my target is different than yours, and I should just go my own way with the firmware. It boils down to preference, but I'll happily collaborate in any way you'd like. Or if you'd rather I didn't use your work as a base, I can start from scratch; I can understand that desire, even in an open-source setting.

Original rambling follows, bearing the former in mind;

I understand that ESPHome makes complex logic difficult. I tried to follow the "easier" logic provided by your sample and frankly it's not what I would call close to intuitive in either form without committing the effort to understand a significant margin more front-loading of "different" than most languages require. I don't "speak" many programming languages, but I can confidently fiddle in most of them, and ESPHome's language is obtuse.

ESPHome seems to be the largest issue, for me personally. OTA, WebUIs, and device libraries readily exist within the Arduino/PlatformIO ecosystems. They're generally implemented with handfuls of bog-standard "typical programming language" schemes, so even non-advanced users can pick through them. OTA with Arduino and/or PlatformIO may be a few steps longer, but it's not what I would describe as harder. In fact, I recently found out that the CLI end of it is shockingly easy, if you just want to build/upload to targets on your home server and code on your desktop, for example.

ESPHome certainly has its offerings, but it requires installing an additional (permanent) service and makes anything beyond dead-simple logic, frankly, obtuse. To me personally, ESPHome is a no-go. Seems perfect for more simplified projects like lights and switches and relays and sensors, or the DIY-tweaker's madness of combinations thereof. Tasmota does that part in my home, but ESPHome certainly makes sense as an alternative for most of even my own devices. I don't dislike ESPHome, I just don't use/need it, and it doesn't (seem to) simplify more than it complicates in the "smart vent" case -- especially for my personal use so obviously I'm biased.

===

ESPHome preferences aside, a limit switch defines one endpoint. Assemble with the servo at zero and the vent closed, use the limit switch to detect the vent's assembled full-open position on powerup, and you're calibrated. Or vice-versa open/closed, I suppose. I could see that working quite well, easier than the same process with only a servo+feedback but not radically easier.

Given that I've already printed, modified+printed, assembled, and modelled a few accessories for this vent, I'll probably just start my first PlatformIO project and use it sans-switch. The fact that I'm out of filament until next week has nothing to do with it, lol. I suspect I'll need to add one as you've decided, though.

TonyBrobston, 2022-10-27:
> Apologies this took so long, took a little time away. Looking back over what I'm about to send, the big end of it appears to be that my target is different than yours, and I should just go my own [...]

Have a quick look at the notes on the top of the license; the main two I'd point out are "permission to modify" and "condition of using the same license". The gist of picking this license was to keep things open source, but that those that modify, need to offer their work to everyone for free, as I have done; this essentially deters companies that want to close source. As long as you follow the terms of the license, I'm cool with whatever you'd like to do; I welcome a fork with modifications, if that's what you'd like to do. I may merge some of it into the main repository, if it fits.
 
> Original rambling follows, bearing the former in mind;
>
> I understand that ESPHome makes complex logic difficult. I tried to follow the "easier" logic provided by your sample and frankly it's not [...]

This is the issue with abstractions. They simplify things, if you understand the framework and how to configure it. Pretty much every common problem has a framework and there's a reason people learn them. However abstractions are really hard to understand if you don't understand the concept they are abstracting; in this case, C/C++ Arduino code, so that's a good place to start. If you know Arduino well, ESPHome is understandable after 5-10 hours of tinkering. But, if you have no interest in learning ESPHome, that also makes it hard to learn 😉 .

The short explanation to learning ESPHome is to look at each high level component, like `cover`, `sensor`, `binary_sensor`, etc., then search that component in [the docs](https://esphome.io/) and read. I think the next big hurdle of understanding is how components work together.

> ESPHome seems to be the largest issue, for me personally. OTA, WebUIs, and device libraries readily exist within the Arduino/PlatformIO ecosystems. They're generally implemented with handfuls of [...]

I went down this route originally. If all I had to focus on was the coding side of things, I would have probably stuck with Arduino C/C++; but... I have a lot of things to juggle and ESPHome was a clear winner.

> ESPHome certainly has its offerings, but it requires installing an additional (permanent) service and makes anything beyond dead-simple logic, frankly, obtuse. To me personally, ESPHome is a no-go. [...]

That isn't quite correct. ESPHome does not require you to have the ESPHome Dashboard running in order to run an ESPHome device. It does however offer a faster way to update a configuration and compile/upload it to a device. It's possible to use the ESPHome CLI and [this upload tool](https://web.esphome.io/) to compile/upload without the ESPHome Dashboard. However, for users that are less technical, all they will need to do is download a `.bin`, navigate to [this upload tool](https://web.esphome.io/), flash/upload, and then enter their wifi credentials. Assuming they never touch any ESPHome configurations directly, they will never need to use the ESPHome Dashboard or CLI.

Also, I would highlight that at a high level, we just have a difference in opinion; which is fine. If you don't have interest in ESPHome, we can leave it at that. I just wanted to make sure you're well informed.

> ===
>
> ESPHome preferences aside, a limit switch defines one endpoint. Assemble with the servo at zero and the vent closed, use the limit switch to detect the vent's assembled full-open position on [...]

I have the limit switch setup to find full closed. I did this because I want closed to be perfect, while open can be off by a few degrees, as long as it doesn't run into the end stop.

> Given that I've already printed, modified+printed, assembled, and modelled a few accessories for this vent, I'll probably just start my first PlatformIO project and use it sans-switch. The fact that [...]

Sounds good. If you decide to add a limit switch, it should only be a matter of reprinting the lower enclosure and the levers/linkage (unless you're clever and remove the lever that needs replaced and reprint only that).

TonyBrobston, 2022-12-07:
I'm going to go ahead and close this.

Let me know if you have any other questions.

closed 2022-12-07 by TonyBrobston