BrobstonCreations/yet-another-smart-vent #26: Request: 4x14 Vent Design

jametsitsi, 2022-11-30:
I have a handful of registers that are 4x14. Is it possible to get a slightly longer model?

TonyBrobston, 2022-12-03:
Yep. I'll add it to the list in the readme. Not quite sure when I'll get to it, but relatively soon.

TonyBrobston, 2022-12-07:
I'm going to go ahead and close this issue. I've add 4x14 to the list and will get to it when I have time. I'll try to remember to write a comment here when it is done.

Let me know if you have any other questions.

closed 2022-12-07 by TonyBrobston

TonyBrobston, 2022-12-07:
@jametsitsi This now exists: https://www.printables.com/model/334635-yet-another-smart-vent-4x14

If you happen to print one, please upload a picture to Printables. If you run into issues, feel free to comment back here.

jametsitsi, 2022-12-07:
@TonyBrobston Awesome, thanks for creating that! I'll get a picture posted when I print it out!