BrobstonCreations/yet-another-smart-vent #27: Thank you & Question

eddyfos, 2022-12-04:
Thank you so much for designing this! I wanted to use Smart Vents, but my vent size is unsupported by Flair at this time (2x12).

Do you have any advice on what smart thermostats to pair this with/how to do that? I have a Google Nest themostat. If I was using Flair- it appears I get a couple of their thermostats and their system controls the vents. I assume theres some way to connect it all in ESPHome (I havent use that before, I'm more used to SmartThings).

TonyBrobston, 2022-12-07:
> Thank you so much for designing this!

No problem 😄 .

I'm going to answer you questions slightly out of order, but hopefully the response will help.

> I assume theres some way to connect it all in ESPHome (I havent use that before, I'm more used to SmartThings).

To clarify, [ESPHome](https://esphome.io/) as they describe it is, "a system to control your ESP8266/ESP32 by simple yet powerful configuration files and control them remotely through Home Automation systems." So ESPHome is just a tool that allows a developer like me to create a device that becomes a first class citizen of Home Assistant. It may be possible to use an ESPHome device with SmartThings, I would tend to think this would only be possible by using MQTT, if there is some sort of integration for that for SmartThings.

> If I was using Flair- it appears I get a couple of their thermostats and their system controls the vents. 

Flair calls these thermostats "pucks". These are essentially temperature sensors that display the room's temperature and allow you to input a desired temperature for the room.

> Do you have any advice on what smart thermostats to pair this with/how to do that? I have a Google Nest themostat. 

This sort of depends on your system and what is important to you. For example, for myself; I have an Ecobee 4. I like this thermostat because it works through Home Assistant's "HomeKit Controller" integration and through this integration all communication is done locally within my network, so if my internet goes down, the integration does not break.

I think what you are really trying to understand is if this system has feature parity with Flair. The short answer is no, at least not yet. I previously wrote a system that would start to add functionality that can read temperature sensors, adjust vent positions, read room desired temperatures, and manipulate the main thermostat in order to get rooms closer to their desired temperature. That is [this project](https://github.com/BrobstonCreations/mqtt-hvac-vent-control). I would consider this a prototype; soon after writing it, it became clear that docker and mqtt were somewhat cumbersome for users to deal with. Based on this feedback I plan to write a Home Assistant integration, which will simplify the installation of the system. I'm not sure how soon I will work on this, but it's on my list.

So another part of the answer is that if you're on SmartThings and you want to take advantage of this future Home Assistant Integration you'll either have to switch over to Home Assistant or run it in addition to SmartThings.

Let me know if that helps to answer your questions.

TonyBrobston, 2022-12-23:
I'm going to go ahead and close this issue. Feel free to write back if you have more questions.

closed 2022-12-23 by TonyBrobston