BrobstonCreations/yet-another-smart-vent #28: Can Not Open/Close Vent.

craftmaster100, 2022-12-06:
I love the idea of this project, however, I'm not sure what has happened. I can not control the vent to either open or close it. 

I am using the dfrobot-dms-mg.

I started off by flashing the yet-another-smart-vent-dfrobot-dms-mg90-a-factory.bin file to the esp. I installed the lever on the motor vertically doing my best to get it perfect. I proceeded to create a device in esphome, adding in the yaml code + the code specifically for the dfrobot.

`globals:
  - id: position_to_move_servo_to
    type: float
    initial_value: '0'
  - id: servo_distance_to_move
    type: float
    initial_value: '0.01'
  - id: closed_position
    type: float
  - id: closed_potentiometer_position
    type: float
  - id: opened_position
    type: float
  - id: opened_potentiometer_position
    type: float

esp8266:
  board: d1_mini

esphome:
  name: tyler-bedroom-vent
# name_add_mac_suffix: true
  project:
    name: esphome.project-template
    version: '1.0'
  on_boot:
    priority: -100
    then:
    - servo.write:
        id: vent_servo
        level: 0
    - delay: 225ms
    - while:
        condition:
          lambda: 'return !id(closed_limit_switch).state;'
        then:
            - globals.set:
                id: position_to_move_servo_to
                value: !lambda 'return id(position_to_move_servo_to) - id(servo_distance_to_move) * ${invert_value};'
            - servo.write:
                id: vent_servo
                level: !lambda 'return id(position_to_move_servo_to);'
            - delay: 75ms
    - globals.set:
        id: closed_position
        value: !lambda 'return id(position_to_move_servo_to);'
    - globals.set:
        id: closed_potentiometer_position
        value: !lambda 'return id(vent_servo_potentiometer).sample();'
    - delay: 225ms
    - servo.write:
        id: vent_servo
        level: !lambda 'return id(position_to_move_servo_to) + ${ninety_degree_servo_distance} * ${invert_value};'
    - delay: 3s
    - globals.set:
        id: opened_position
        value: !lambda 'return id(position_to_move_servo_to) + ${ninety_degree_servo_distance} * ${invert_value};'
    - globals.set:
        id: opened_potentiometer_position
        value: !lambda 'return id(vent_servo_potentiometer).sample();'

cover:
  - platform: template
    device_class: damper
    id: vent
    name: Vent
    open_action:
      - servo.write:
          id: vent_servo
          level: !lambda 'return id(opened_position);'
    close_action:
      - servo.write:
          id: vent_servo
          level: !lambda 'return id(closed_position);'
    stop_action:
      - servo.write:
          id: vent_servo
          level: !lambda 'return remap(id(vent_servo_potentiometer).state, id(opened_potentiometer_position), id(closed_potentiometer_position), id(opened_position), id(closed_position));'
    tilt_action:
      - servo.write:
          id: vent_servo
          level: !lambda 'return remap(tilt, float(1), float(0), id(opened_position), id(closed_position));'
    tilt_lambda: !lambda 'return remap(id(vent_servo_potentiometer).state, id(opened_potentiometer_position), id(closed_potentiometer_position), float(1), float(0));'
    lambda: 'return id(closed_limit_switch).state ? COVER_CLOSED : COVER_OPEN;'

binary_sensor:
  - platform: gpio
    pin: 
      number: 16
      mode:
        input: true
        pulldown: true
    id: closed_limit_switch
  
sensor:
  - platform: adc
    id: vent_servo_potentiometer
    pin: A0
    update_interval: 250ms

servo:
  - id: vent_servo
    output: vent_servo_output
    transition_length: ${servo_transition_length}
    auto_detach_time: 200ms
    
substitutions:
  ninety_degree_servo_distance: '0.77'
  servo_transition_length: 7s
  invert_value: '1'
    
output:
  - platform: esp8266_pwm
    id: vent_servo_output
    pin: D3
    frequency: 50 Hz

captive_portal:
    
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:

api:

mdns:
    
ota:
  password: "39434a9d48f484dcea835f2515f6bda5"
    
logger:
  level: INFO`

This is the exact code I flashed to the esp. Everything looks good in the logs.

`Using 'COM4' as serial port.
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP894 %)Writing at 0x00067b38... (100 %)Wrote 441264 bytes (305707 compressed) at 0x00000000 in 10.1 seconds (effective 349.8 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[15:35:28][I][logger:258]: Log initialized
[15:35:28][I][app:029]: Running through setup()...
[15:35:34][I][wifi:444]: - 'iot network' ([xxxxxxxxx](xx:xx:xx:xx:xx:xx)) ▂▄▆█
[15:35:34][I][wifi:444]: - 'iot network' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[15:35:34][I][wifi:257]: WiFi Connecting to 'iot network'...
[15:35:36][I][wifi:518]: WiFi Connected!
[15:35:36][W][ota:102]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[15:35:36][I][app:062]: setup() finished successfully!
[15:35:36][I][app:102]: ESPHome version 2022.11.3 compiled on Dec  6 2022, 15:20:09
[15:35:36][I][app:104]: Project esphome.project-template version 1.0
[15:35:36][W][ota:102]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[15:40:28][I][ota:113]: Boot seems successful, resetting boot loop counter.`

The device is connected to my network. I verified by checking the router. I can access the device via its IP address. When I plug the device in it moves to the open position, then tilts a couple of degrees slightly. When I check the device's state on my browser it says it is open. The controls, however, are unresponsive. Furthermore, when I add the device in the integrations tab in home assistant, it shows up and appears to be functioning just fine. When you send an open or close command I get an error saying:

Failed to call service cover/close_cover. Connection not done for tyler-bedroom-vent @ 10.109.1.149!

I can verify home assistant / esphome integration is working fine as I have other esp devices that work just fine.

Any ideas are appreciated, thanks.

TonyBrobston, 2022-12-07:
> I started off by flashing the yet-another-smart-vent-dfrobot-dms-mg90-a-factory.bin file to the esp. I installed the lever on the motor vertically doing my best to get it perfect. I proceeded to create a device in esphome, adding in the yaml code + the code specifically for the dfrobot.

I need to review the documentation, as I was looking for this section it seemed a little confusing to me; but it's late in the day. Did you do the fourth note here?: https://github.com/BrobstonCreations/yet-another-smart-vent/blob/97521b56b3b9ac1244692dcf15246cff5b8c8766/docs/VENT_ASSEMBLY.md#notes
I'm not sure if you mixed up the two `.bin` file names? But the `turn-to-middle.bin` should be flashed, then the servo wiper arm should be attached.

This may or may not be the issue, but I'm trying to start from the earlier steps and confirm that they are right before speculating about the later stuff.

TonyBrobston, 2022-12-07:
Also to clarify. On boot the vent should turn until it depresses the limit switch, then it should turn roughly 90 degrees, and then should be rear for use. My initial guess is that the wiper arm isn't aligned close to the middle of the range. It's entirely possible that I'm wrong, but that's where I'd start.

craftmaster100, 2022-12-09:
Hey @TonyBrobston,

Thanks for the reply. You are right, I accidentally flashed the wrong file in order to position the motor in the center. I went ahead and re-flashed the correct file this time. Reassembled the motor arm and flashed the code. I was getting the same thing as before, appeared to be functioning but I still had no control over the motor. 

So I dived in a little deeper, starting over. Turns out the real source of my problem was pins A0 and D0 were connected by a tiny bit of solder. Rookie mistake. I should have checked everything with a multimeter, luckily nothing shorted out. Threw everything back together and flashed the code again. Works like a charm now.

Thanks for designing this cool little gadget. I got one room in my house that's always hotter than blazes because of a server running in there. This vent is perfect so it can automatically be closed when I am heating the house.

TonyBrobston, 2022-12-09:
@craftmaster100 No problem. Glad you got it to work. If you happen to have a minute and could upload a make to Printables, that would help me out a lot. No biggie if not.