BrobstonCreations/yet-another-smart-vent #29: how do I upload the file.

bob5731, 2022-12-12:
How do I upload the file to home assistant?

TonyBrobston, 2022-12-12:
I'm not quite sure what you're asking.

It looks like you grabbed one of the yaml files from the root of this project and tried to do something weird with it.

Have you read through the documentation?

bob5731, 2022-12-12:
ware we the code to connect to home assistant. your link does not work. in setup step.

TonyBrobston, 2022-12-12:
Can you link me to where you've gotten to in the documentation so I can get some context?

bob5731, 2022-12-12:
https://github.com/BrobstonCreations/yet-another-smart-vent/blob/master/docs/COMPILE_AND_UPLOAD.md

bob5731, 2022-12-12:
ESPHome dashboard.

TonyBrobston, 2022-12-12:
I haven't spent a lot of time using ESPHome Dashboard lately. But, you should be able to merge the yaml file that correlates the servo you're using with the included file (https://github.com/BrobstonCreations/yet-another-smart-vent/blob/master/common/yet-another-smart-vent-base.yaml) to make one file. I think all that should take is copying the substitutions section and moving that to the top of the included file and then copy/paste that whole file into your ESPHome Dashboard.

In order to make maintenance easier for myself and because I'm supporting two stepper motor options, I am using an include of another yaml file. This enables me to keep things that are common between the two configurations in one file.

bob5731, 2022-12-12:
I got your code to show up on my D1 mini using the included in file. Now how do I set it up in home assistant?

TonyBrobston, 2022-12-12:
I would tend to think ESPHome would function to some extent without having the servo or limit switch wired up, but I can't say for sure, since I haven't tested that.

Are you able to navigate to the D1 Mini's IP Address?

Do you have another D1 Mini that you could test with also to see if you're getting the same behavior?

All the D1's I've flashed with one of my bin files has either showed up immediately in Home Assistant or I had to manually enter the D1's IP Address into the ESPHome Integration. In basically all cases, I was able to control the D1 through Home Assistant. Though, I've had D1 quality issues; anything from being dead on arrival, to mis-wiring accidently and frying the chip, to faulty soldering from the factory that also fried things.

Have you tested your D1 Mini with any other ESPHome configurations? That might help narrow down the issue.

bob5731, 2022-12-12:
no I have not tested the d1 mini ESPHome. But can navigate to the D1 Mini's IP Address. I have on hand micro servo s51's.

TonyBrobston, 2022-12-12:
You can also text things without printing any parts. I have done this a handful of times.

I'd be curious to see what shows up when you hit the D1 Mini's IP Address. Is that where you're getting the "connection closed" logs from?

bob5731, 2022-12-12:
yes  that where i'm getting the "connection closed" logs from

TonyBrobston, 2022-12-12:
I'm not sure what else to suggest 🤷‍♂️.

bob5731, 2022-12-13:
the bug is in your code!!! I need the it fixed so. I can use my air vent's. Can you add micro servo 9g s51 to air vent's?

TonyBrobston, 2022-12-13:
I don't appreciate the accusation.

It's entirely possible that there is a bug. Feel free to open a Pull Request to fix whatever you think it is.

No I will not be adding support for the SG90. I will not recommend or support a plastic gear servo.

I have no obligation to do anything for you. Feel free to be more kind, appreciative, and helpful; then I will consider reciprocating.

jametsitsi, 2022-12-14:
@bob5731 There's a good chance you're running into issues with the SG90 because there is no 4th wire for analog feedback (see the adc on pin A0 https://github.com/BrobstonCreations/yet-another-smart-vent/blob/1.1.0/common/yet-another-smart-vent-base.yaml#L95..L99). Sounds like you may need to get more familiar with the yaml. What you see in Home Assistant isn't going to mean/do much without a servo wired up. What are you expecting to see in Home Assistant without the servo and limit switch?
@TonyBrobston did some good work here and works as expected. I initially tested a few things out with the SG90 while I was waiting for my servos to come in. There's only so much you can do - getting a servo with analog feedback is going to be best.

TonyBrobston, 2022-12-23:
> @bob5731 There's a good chance you're running into issues with the SG90 because you dont have a limit switch. One way to test this out is to use a wire from 3.3V to GPIO 16 to trigger a closed position. Also something to note: there is no 4th wire for analog feedback (see the adc on pin A0 https://github.com/BrobstonCreations/yet-another-smart-vent/blob/1.1.0/common/yet-another-smart-vent-base.yaml#L95..L99). Sounds like you may need to get more familiar with the yaml. What you see in Home Assistant isn't going to mean/do much without a servo wired up. What are you expecting to see in Home Assistant without the servo and limit switch? @TonyBrobston did some good work here and works as expected. I initially tested a few things out with the SG90 while I was waiting for my servos to come in. There's only so much you can do - getting a servo with analog feedback and metal gears and a limit switch is going to be best.

Thanks for the input @jametsitsi . I agree with everything you said. It is possible to open an SG90 and add a wire to it's potentiometer, I've tested this and it works; but again, I won't recommend or support a plastic gear servo.

I'm going to go ahead and close this issue.

closed 2022-12-23 by TonyBrobston