BrobstonCreations/yet-another-smart-vent #30: Servo options

NavinF, 2022-12-24:
>https://brobston-creations.mybigcommerce.com/yet-another-smart-vent-electronics-and-fasteners-kit/

What's the difference between the two servos? (DFRobot DMS-MG90-A and Batan B2122)

The MG90 looks like a smaller micro servo so I'm guessing the other has a bit more oomph, but does this matter at normal airflow rates?

TonyBrobston, 2022-12-24:
Here are the specs for both:

DMS-MG90-A:
https://www.dfrobot.com/product-1970.html
[image]

Batan B2122:
https://www.adafruit.com/product/1450
[image]

I'm not super experienced with servos, so take this with a grain of salt. In general, if the servo can get the job done, I'm not reading too far into the specs. I've tested both servos with the 4x12 variation and will soon test the DMS-MG90-A on the 6x10. As the vent sizes get bigger, I do wonder if the servo will have enough power to turn the louvers; some day if I build a 12x12, I definitely plan to experiment with two servos, each controlling half the louvers.

When comparing the two servos it's hard for me to say what is important and what isn't. I noticed the DMS-MG90-A has a stall torque of "1.5±0.1kg·cm @6.0V" and the Batan B2122 has torque of "25 oz*in / 1.8 kg*cm @6V". I'm not sure if "stall torque" and "torque" are the same thing in this context; but if they are, 1.5 is less than 1.8, so I think it's fair to say that the Batan B2122 has more oomph. I also noticed the Batan B2122 lists "Dual ball bearing", while the DMS-MG90-A says nothing about bearings; it is unclear to me if the DMS-MG90-A has any bearings or not; again, not sure if this detail is important. I've noticed that the Batan B2122 is slightly quieter, but not enough to matter to me personally.

I think the biggest notable difference is price. Depending on how many you order and from where, you can get them for slightly less, but in general the Batan B2122 is about twice as expensive as the DMS-MG90-A while (in my opinion) the Batan B2122 is not offering twice the value; maybe like 10% better than the DMS-MG90-A (again, in my opinion).

Does that help answer your question?

TonyBrobston, 2022-12-24:
Also the two servos are comparable in size; I would say the Batan is slightly smaller, but not by much.

NavinF, 2022-12-24:
@TonyBrobston Thanks that was really helpful! I went ahead and ordered 3 kits with the B2122 version. I totally forgot about noise, but for my use case even a small reduction would make quite a difference :)

I intend to cycle these maybe every 10 minutes so I can use a central unit to heat one room and recirculate air for the other two. Dunno if that'll work, but I know that off the shelf smart vents have zero chance of solving problems like this. 

btw you might wanna add this comparison (same size, slightly less noise, more torque and whether that makes a difference) to the product page. A lot of people abandon their carts when they can't decide.

closed 2022-12-24 by NavinF

TonyBrobston, 2023-01-01:
> @TonyBrobston Thanks that was really helpful! I went ahead and ordered 3 kits with the B2122 version. I totally forgot about noise, but for my use case even a small reduction would make quite a [...]

@NavinF I'll be curious to hear about what results you get. I think what you're saying is doable, but I don't know exactly how your system is set up. If you run into any issues, let me know. But yes, off the shelf smart vents are pretty closed off. I'm definitely a fan of things being able to be hacked on.