BrobstonCreations/yet-another-smart-vent #32: Limit switch location?

PInnovation, 2023-05-05:
I'm using the DMS servo and the limit switch soldered to only the D1 mini power shield (D0 and 3v3), but I'm struggling to figure out where to attach the limit switch. I can't seem to find any pictures anywhere.

I've tried operating it unassembled to test the electronics setup (on two separate units) and, after I upload yet-another-smart-vent-dfrobot-dms-mg90-a.yaml, the vent snaps to one position then gradually moves about 120 degrees, but then I go to ESPHome on HA and "open cover" is grayed out, but "close cover" does nothing. If I try to manually depress the limit switch before it gets to 120 degrees, it doesn't seem to have any effect.

What am I missing?

TonyBrobston, 2023-05-06:
@PInnovation 

I probably need to go through the docs one more time and make sure everything is as TLDR as possible.

Did you happen to do the step where you flash the `turn-to-middle-factory.bin` and install the wiper on the servo after the program starts? If not, the fourth bullet here describes how to: https://github.com/BrobstonCreations/yet-another-smart-vent/blob/master/docs/VENT_ASSEMBLY.md#notes

The limit switch goes here: 
[image]
The hole on the right is for the hole on the limit switch and the indentation on the left is to help keep it aligned.

The point in development where I added the limit switch came after the documentation. Unfortunately the time and effort to do the work and keep it all up to date and not make much if any money, is difficult.

Let me know if this helps, I'm sure we can get things figured out.

TonyBrobston, 2023-05-06:
It might also be worth mentioning that at startup it runs this bit: https://github.com/BrobstonCreations/yet-another-smart-vent/blob/02c842eb81c6238af2f971708d5a1ac2aa1541fa/common/yet-another-smart-vent-base.yaml#L29-L60 which tells the program to turn the servo until the limit switch is pressed, then record the potentiometer position, then turn the servo a hardcoded distance (roughly 90 degrees) and then record the potentiometer position again. If the wiper isn't attached at roughly the center of the servo's range of motion or the limit switch isn't pressed before the servo hits its max range of motion, things won't work correctly.

PInnovation, 2023-05-12:
@TonyBrobston It turns out I had missed an important step. I read through the Electronics Assembly and hadn't realized that the limit switch had been added later to the project. I read step 18 and attached the limit switch to 3v3 and D0 of the power board, and briefly thought it was odd that it wasn't connected to the D1 Mini as well, but assumed I was following the directions properly and that there was some circuitry in the power shield that I wasn't aware of. I now see that step 1 calls out attaching the header pins to 3v3 and D0 - my bad.

Even after fixing that, the two assemblies that I had tested without the header pins had some erratic motor behavior. They would stutter a lot and wouldn't completely open the louvers. It seemed like a damaged gearing or internal sensor issue. I replaced the motor on both of those and they are working well now. Thanks for your help.

closed 2023-05-12 by PInnovation