BrobstonCreations/yet-another-smart-vent #33: 12x12 vent

sim4dim, 2023-12-05:
I had seen this was asked last year, just wonder if there is any progress on that before I will dive into trying to take a swing at it.

hllhll, 2024-05-19:
I don't see the owner of this repo replying to anyone, lately :( Do u have some progress in this area? I need a 20x40 CM

TonyBrobston, 2024-07-04:
Sorry for the delay on this reply.

I have not investigated this yet. It should be possible. It may require two servos rather than one. The louvers, levers, linkage, etc. care probably be shared or at least scaled. I think the biggest change will be with the electronics enclosure part of the vent shell, if two servos are used. I would likely start by designing a 12x12 with one servo and see whether the servo has enough torque to turn all the louvers; but my guess is that it won't. There are likely some other options, like a stronger servo or something of that nature.

I don't have a timeline on this.